react-native-camera/react-native-camera #956: Android Permissions

ryanvanderpol, 2017-10-23:
I'm a little unfamiliar with how Android permissions work, so forgive me if there is an obvious answer to this. I've followed the installation instructions and got this working fine on our iOS app but the Android version does not prompt for Camera permissions on install.  My `AndroidManifest.xml` has the following lines in it:

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.myapp"
    android:versionCode="1"
    android:versionName="1.0">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-feature android:name="android.hardware.camera" />
    <uses-feature android:name="android.hardware.camera.autofocus" />

    <uses-sdk
        android:minSdkVersion="16"
        android:targetSdkVersion="23" />

....

```

However, when I install the app via the apk, the "do you want to install this app?" screen says it requires no special permissions.  And then once installed, if I go to Settings > Applications > MyApp > Permissions, the Camera permission is not enabled.  Once I enable it manually and open my app I get the following error:

```
TypeError: _reactNativeCamera2.default.checkDeviceAuthorizationStatus is not a function
```

This happens in a block that looks like this:

```
        if(Camera){
            Camera.checkDeviceAuthorizationStatus()
            .then(access => {
                if(!access) {
                    Alert.alert('Oops!', 'We need permission to access your camera to scan codes')
                } else {
                    navigate('Scanner');
                }
            })
        }
```

Like I said above, this exact code works just fine on iOS, so my gut says there is something weird going on with the Android permissions but I can't tell what I'm doing wrong.  Any ideas?

ryanvanderpol, 2017-10-23:
I think we've figured out that apparently the `checkDeviceAuthorizationStatus` API doesn't work for Android and that we need to use `PermissionsAndroid` instead.  This is NOT clear in the README so I will leave this open for now in the hope that someone will update the docs so others won't waste as much time as we did trying to figure this out.

For now, we've changed our code to do the following:

```
        if(Platform.OS === 'android'){
            PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.CAMERA)
            .then(res => {
                if(res !== 'granted') {
                    Alert.alert('Oops!', 'We need permission to access your camera to scan codes')
                } 
                else navigate('Scanner');
            });
        } else {
            if(Camera){
                Camera.checkDeviceAuthorizationStatus()
                .then(access => {
                    if(!access) {
                        Alert.alert('Oops!', 'We need permission to access your camera to scan codes')
                    } 
                    else navigate('Scanner');
                });
            }
        }

```

ptomasroos, 2017-10-24:
Feel free to send a PR making it clearer in the documentation.

closed 2017-10-24 by ptomasroos

pavanmehta91, 2017-11-16:
Where to put the above code? Sorry for the noob question.

dancomanlive, 2018-02-13:
ComponentDidMount