react-native-camera/react-native-camera #959: How to mute video capture start/stop sound on Android?

mihaimartalogu, 2017-10-25:
Hi,

I've been struggling to find a way to disable the annoying sound played at start/stop of a video capture on a Honor 9 phone (Android Nougat). It happens even if I select mute in the stock camera app's menu, and even when the phone is in silent or vibrate mode.
I did not have this issue on my older Galaxy S4 mini (Android Lollipop).

I have tried, without success:
1. Passing `playSoundOnCapture={false}`
2. Disabling shutter sound
```java
if (android.os.Build.VERSION.SDK_INT>=android.os.Build.VERSION_CODES.JELLY_BEAN_MR1){
    mCamera.enableShutterSound(false);
}
else{
    AudioManager audio= (AudioManager)_reactContext.getApplicationContext().getSystemService(Context.AUDIO_SERVICE);
    audio.setStreamVolume(AudioManager.STREAM_SYSTEM, 0,   AudioManager.FLAG_REMOVE_SOUND_AND_VIBRATE);
}
```
3. Muting system streams
https://github.com/lwansbrough/react-native-camera/pull/583

Note that this start/stop sound is NOT played by the stock camera app and other apps like Snapchat or the sample video app https://github.com/googlesamples/android-Camera2Video

Could it be something related to the fact that react-native-camera uses the Camera1 API whereas these other apps use the Camera2 API?

salimbutex, 2017-12-27:
hi man did u get any solution yet? 
playSoundOnCapture={false}

this never works

mihaimartalogu, 2017-12-27:
Hi, no, it's frustrating but I ended up rolling my own React Native component using Camera2 API in order to work around this issue...

Friendly-Robot, 2018-05-03:
Thanks @mihaimartalogu for pointing out the root of the problem. I looked into `RNCamera.js` and noticed that there is a prop called `useCamera2Api` that isn't noted in the docs. Set this prop to true and `playSoundOnCapture={false}` should work as you'd discovered.

IssuehuntBot, 2018-12-03:
@issuehuntfest has funded $20.00 to this issue. [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/959)

closed 2019-06-08 by stale[bot]

Martlohmus12, 2020-01-07:
I had the same problem with muting start capture sound. You can mute the starting sound with (playSoundOnCapture={false}) when camera2api is enabled, but before camera2api wasn’t 
being honoured. My pullrequest was merged (https://github.com/react-native-community/react-native-camera/pull/2603), which enables to use camera2apis features, Including playSoundOnCapture.