react-native-camera/react-native-camera #960: app crashes while opening camera

goel135, 2017-10-26:
i have tries by changing the version also but the problem is still persisting

jgcmarins, 2017-10-26:
@goel135 could you provide more details?

goel135, 2017-10-26:
@jgcmarins 
    "react-native-camera": "^0.10.0",
    "react-native": "0.46.4",
    "react-navigation": "^1.0.0-beta.11",

	i have already added permissions 
 <uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-feature android:name="android.hardware.camera" />
    <uses-feature android:name="android.hardware.camera.autofocus" />
    <uses-permission android:name="android.permission.READ_PHONE_STATE" />

    compileSdkVersion 25
    buildToolsVersion "25.0.2"

Do tell me if anything else i should provide you

jgcmarins, 2017-10-26:
I guess it's ok.
Are you running it on simulator or device? Android? iOS?

goel135, 2017-10-26:
Android device
Android version 6.0.1

jgcmarins, 2017-10-26:
could you try to run `adb logcat` and paste the error message here?

LinFeiLong, 2017-10-26:
Hello, I'm actually having the same issue, but on iOS.

`<Camera
            captureQuality={Camera.constants.CaptureQuality.medium}
            captureTarget={Camera.constants.CaptureTarget.disk}
            type={Camera.constants.Type.front}
            ref={(cam) => {
              this.camera = cam
            }}
            style={[styles.pictureContainer, { height: Metrics.width, width: Metrics.width }]}
            aspect={Camera.constants.Aspect.fill} />
`

The app crashes if I'm using the component.

goel135, 2017-10-26:
I am getting null pointer exception and camera service failed error.

goel135, 2017-10-26:
java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference
at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent(RCTCameraViewFinder.java:349)
at android.view.View.dispatchTouchEvent(View.java:9306)
at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2198)
at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2198)
at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2198)
at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2198)
at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2198)
at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2198)
at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2198)
at com.android.internal.policy.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2403)
at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1737)
at android.app.Activity.dispatchTouchEvent(Activity.java:2820)
at com.android.internal.policy.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2364)
at android.view.View.dispatchPointerEvent(View.java:9526)
at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4267)
at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4133)
at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3674)
at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3727)
at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3693)
at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:3819)
at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3701)
at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:3876)
at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3674)
at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3727)
at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3693)
at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3701)
at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3674)
at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:5959)
at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:5933)
at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:5894)
at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6062)
at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:187)
at android.os.MessageQueue.nativePollOnce(Native Method)
at android.os.MessageQueue.next(MessageQueue.java:323)
at android.os.Looper.loop(Looper.java:138)
at android.app.ActivityThread.main(ActivityThread.java:5527)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:739)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:629)

kraffslol, 2017-10-30:
If you're running a recent version of Android you will need to make sure the camera has permissions before using it. Try requesting the camera permission before showing it. See [PermissionsAndroid](https://facebook.github.io/react-native/docs/permissionsandroid.html)

LinFeiLong, 2017-10-31:
@kraffslol 
I tried on my issue on iOS and it works. Thx.

goel135, 2017-11-01:
@kraffslol 
I have also tried it on lower versions of android but the problem was same.
No solution found yet.

shubhangi-asgola, 2017-11-06:
Did anyone find any solution for this? I am facing the same problem in android device. 
When I run it on simulator it gives me following error:

Attempt to invoke virtual method 'android.hardware.Cameracom.Iwansbrough.RCTCamera.RCTCamera.acquireCameraInstance(int)' on a null object reference

hasFlash
RCTCameraModule.java:663

goel135, 2017-11-08:
Any update on this issue ??

shubhangi-asgola, 2017-11-08:
@goel135 I ended up using the below configurations:

React Native version: 0.41.2
react-native-camera version: 0.6

Even after that I was facing problems with camera instantiation. For now I am checking inside hasFlash() of RCTCameraModule.java if the camera instance is not present then create one. I took help from a java guy.
There should be some elegant solution for this. For now I am using this. Hope this helps.

ospfranco, 2018-01-12:
I think this had to do with android 6.0+ and how permissions were being handled (from 6.0 you need to prompt for individual permissions).

This should be fixed on master.

closed 2018-01-12 by ospfranco

rizalmalik0, 2018-01-23:
+1

React Native version: 0.42.3
react-native-camera version: 0.6

same issue on this, i think this issue happen on some devices with 3rd party permission

ortonomy, 2018-09-14:
I'm still having this isssue on iOS. It just crashed. Any thoughts?