react-native-camera/react-native-camera #963: Flash mode doesn't work for Android

Macstyg, 2017-10-31:
I've tried `flashMode={Camera.constants.FlashMode.auto}` and `flashMode={Camera.constants.FlashMode.on}` and both have no any effect on camera. 
But the `torchMode` works as expected.
Device: LG g6, android 7
```
"react-native": "0.46.3"
"react-native-camera": "0.10.0"
```

oneenam, 2017-11-23:
also for iOS doesn't work :(
"react-native": "0.44.3",
"react-native-camera": "^0.12.0",

hiennguyen92, 2017-11-28:
Flash only light up when shooting, not always

ACCTFORGH, 2017-12-24:
I found where the error is. The setFlashMode in RCTCameraView.java is responsible for setting the 
flashmode prop. However, it relies on the  _viewFinder argument to initialized to work, which was done via the type prop setting as shown in the code below. Because type and flashMode prop are handled almost at the same time by React Native, when flashMode is getting set, the _viewFinder is still null and the prop setting  can't happen as shown in Snippet 2. 

The hack way to solve this issue is to hook the flashMode prop to a state in JS and set it again 1/2s after component mound. The best way to solve this error and the rest of the prop setting issue will be to shift these to the public methods such as capture.

Snippet 1:

    public void setCameraType(final int type) {
        if (null != this._viewFinder) {
            this._viewFinder.setCameraType(type);
            RCTCamera.getInstance().adjustPreviewLayout(type);
        } else {
            _viewFinder = new RCTCameraViewFinder(_context, type);
            if (-1 != this._flashMode) {
                _viewFinder.setFlashMode(this._flashMode);
            }
            if (-1 != this._torchMode) {
                _viewFinder.setTorchMode(this._torchMode);
            }
            addView(_viewFinder);
        }
    }

Snippet 2: 
    public void setFlashMode(int flashMode) {
        this._flashMode = flashMode;
        Log.i("RCTCameraView", "setFlashMode: " + (this._viewFinder != null)); //prints false
        if (this._viewFinder != null) {
            this._viewFinder.setFlashMode(flashMode);
        }
    }

jgcmarins, 2017-12-26:
@ACCTFORGH PRs are welcome 😄

ACCTFORGH, 2017-12-26:
shifting props to the capture method doesn't seem to reliably work and it's costly to set the camera parameters for each capture. Quickest hack solution now is in JS the update state a bit time after the camera component mounts to make the prop values take effect.

mcanobbio, 2018-01-25:
Hi I'm usign react-native 0.51.0 and The hack provided by @ACCTFORGH works fine with the version
react-native-camera 0.12.0

but doesn't work when I use
react-native-camera in master 

And neither with 
react-native-camera 0.13.0

sibelius, 2018-01-25:
can u try RNCamera?

mcanobbio, 2018-02-01:
@sibelius sorry, but what do you mean with try RNCamera?

sibelius, 2018-02-01:
https://github.com/react-native-community/react-native-camera#experimental

m-idler, 2018-02-02:
Works with `RNCamera` and the latest master neither 😕

mcanobbio, 2018-02-12:
@sibelius still doesn't work with RNCamera

m-idler, 2018-02-14:
i got it working with ios but not with android so far. 
@mcanobbio does it work for you with both, ios and android? How do you currently implement it?

mcanobbio, 2018-02-16:
@m-idler Yes works on both platforms.

This is how implement it

`<RNCamera
            ref={(cam: any) => {
              this.camera = cam
            }}
            style={styles.camera}
            flashMode={RNCamera.Constants.FlashMode.on}
            captureQuality={'high'}
          />`

are you sure you follow this documentation https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md ?

sibelius, 2018-02-16:
Tks @mcanobbio

closed 2018-02-16 by sibelius

raphaelhovsepyan, 2018-09-06:
@Macstyg  Maybe you need `flashMode={RNCamera.Constants.FlashMode.torch}`

tajpouria, 2019-05-27:
Flash only light up when u try capture 

```javascript

const {
  FlashMode: { on, off },
  Type: { back, front }
} = Camera.Constants;

  this.state = {
      capturing: false,
      hasCameraAndAudioPermission: null,
      captures: [],
      flashMode: on,
      cameraType: back
    };

 <Camera type={cameraType} flashMode={flashMode} style={styles.preview} />

```