react-native-camera/react-native-camera #964: Camera crashing app

BrentonWatt, 2017-11-01:
Both the front and back facing cameras cause the app to crash on Android, the front facing causes the crash to occur sooner. Barcode scanning is enabled. 

The logcat error is:

**11-01 13:28:39.241 3465-3472/com.tabletcheckin E/dalvikvm: Unexpected opcode(471) with kInstrCanBranch set
11-01 13:28:39.241 3465-3472/com.tabletcheckin E/dalvikvm: VM aborting
11-01 13:28:39.241 3465-3472/com.tabletcheckin A/libc: Fatal signal 6 (SIGABRT) at 0x00000d89 (code=-6), thread 3472 (Compiler)**

Android version is 4.4.4.

Any ideas?

DeividVeloso, 2017-11-01:
Me too, someone else?

aspidvip, 2017-11-08:
yes me to

koswarabilly, 2017-11-20:
My application crash when i try to pass location metadata to the image captured, anyone have the same issue? or yours just crash ?

closed 2019-06-08 by stale[bot]