react-native-camera/react-native-camera #966: App stops working after running react-native-camera install command

Mate38, 2017-11-03:
My app was running perfectly, it was just running the commands `npm install react-native-camera --save` and `react-native link react-native-camera` which started to display the error below:

[image]

I created an app from scratch just to test, ran perfect, after running the commands I received the same error.

jgcmarins, 2017-11-03:
Hello @Mate38 , `react-native link` can sometimes not link well.
Did you followed README instructions?
How did you solved?

Mate38, 2017-11-06:
I checked and the link created everything exactly as described for the manual process, except the permissions, but even adding these manually continues all the same.

ivancho2, 2017-11-16:
i have a equals error. helpm'

Mate38, 2017-11-20:
I tested it again, I created an app from scratch with` react-native init`, the only thing I did differently is that after `npm install react-native-camera --save` I delete the folder `node_modules`, gave an `npm install`, and only then the `react-native link react-native-camera`, and worked everything correctly.

My app.js:

```
import React, { Component } from 'react';
import {
  Dimensions,
  Platform,
  TouchableHighlight,
  StyleSheet,
  Text,
  View
} from 'react-native';

import Camera from 'react-native-camera';

export default class App extends Component {

  takePicture() {
    const options = {};
    //options.location = ...
    this.camera.capture({metadata: options})
      .then((data) => console.log(data))
      .catch(err => console.error(err));
  }

  render() {
    return (
      <View style={styles.container}>
        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          aspect={Camera.constants.Aspect.fill}>
          <Text style={styles.capture} onPress={this.takePicture.bind(this)}>[CAPTURE]</Text>
        </Camera>
      </View>
    );
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: 'row',
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center'
  },
  capture: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    color: '#000',
    padding: 10,
    margin: 40
  }
});
```

closed 2017-11-20 by Mate38