react-native-camera/react-native-camera #969: Camera-focus crash on React Native 0.50.1

richardgirges, 2017-11-05:
Hey there react-native-camera team. Thanks for such an amazing package, it's done wonders for us!

We've recently upgraded to React Native 0.50.1 and have started getting an app crash when attempting to focus using react-native-camera: `Unsupported top level event type "topFocusChanged" dispatched`.

Full stack trace:
```
Unhandled JS Exception: Unsupported top level event type "topFocusChanged" dispatched
reactConsoleErrorHandler @ ExceptionsManager.js:73
console.error @ YellowBox.js:71
logToConsole @ RCTLog.js:48
logIfNoNativeHook @ RCTLog.js:31
__callFunction @ MessageQueue.js:302
(anonymous) @ MessageQueue.js:116
__guard @ MessageQueue.js:265
callFunctionReturnFlushedQueue @ MessageQueue.js:115
(anonymous) @ debuggerWorker.js:72
```

[image]

Here is our `react-native-camera` component code:
```javascript
      <RNCamera
        ref={c => this.camera = c}
        style={{ flex: 1 }}
        deafultOnFocusComponent={true}
        onFocusChanged={e => console.log('FOCUS')}
        keepAwake />
```

package.json:
```
"react-native": "0.50.1",
"react-native-camera": "^0.10.0"
```

Unclear if this is an issue on the react-native-camera side of things or on the RN side of things. Any help would be greatly appreciated!

cesarfigueroa, 2017-11-06:
We’re experiencing this issue as well on the exact same versions of `react-native` and `react-native-camera`

ptomasroos, 2017-11-07:
This is because a breaking change in react-native

richardgirges, 2017-11-12:
Appears to be fixed on react-native 0.50.3 and react-native-camera 0.12.0.

closed 2017-11-12 by richardgirges