react-native-camera/react-native-camera #970: react-native link broken with RN 0.50.1

aforty, 2017-11-06:
Not sure if this si the only version affected but with RN 0.50.1 when I run `react-native link react-native-camera` I get an error:

```
$ react-native link react-native-camera
Scanning folders for symlinks in /Users/aforty/<my-project>/node_modules (12ms)
rnpm-install info Android module react-native-camera is already linked
rnpm-install info Linking react-native-camera ios dependency
rnpm-install ERR! Something went wrong while linking. Error: Cannot read property 'match' of undefined
Please file an issue here: https://github.com/facebook/react-native/issues

Cannot read property 'match' of undefined
```

sibelius, 2017-12-09:
this is a bug on react-native link

closed 2017-12-09 by sibelius

isnifer, 2018-02-12:
@sibelius so, but it doesn't work only with your package, oook

Andreyco, 2018-02-14:
When linking and using Pods, I cannot see RNCamera in targets. Investigating.