react-native-camera/react-native-camera #979: What it there is no info.plist in React-Native project?

Hiroki1116, 2017-11-13:
For iOS 10+, it's  one of the requirements to put the "Privacy - Camera Usage Description" key into info.plist.

However, there is no such a file in my project file, maybe because I made this project with `create-react-native-app` command.
In which file should I put this information?

I'll apologise if this question is too basic. I've just started to use React-Native.
Any advice will be appreciated.

jgcmarins, 2017-11-13:
Hello @Hiroki1116. In your case, as you're using CRNA, you should use Expo API Components. You can find an example here: https://github.com/expo/native-component-list/blob/master/screens/ExpoApisScreen.js#L143

Hiroki1116, 2017-11-17:
@jgcmarins 
Thank you for your reply.
Could you elaborate on how Expo API can be a substitution of info.plist?
(Does that mean that there is an equivalent of info.plist among Expo API components?)
I've been reading Expo's documentation, but I'm still clueless.

giovannipds, 2018-06-18:
Still clueless. Did you learn it, @Hiroki111?

closed 2019-06-08 by stale[bot]