react-native-camera/react-native-camera #983: Camera is transparent into an android modal

l1br3, 2017-11-14:
I've got the `<Camera/>` component (right from exemple in readme) into a modal and it is half transparent. It means that I can see the modal content behind the camera view as if a white opacity of 0.5 was applied.  
This does not happen into a non-modal view.

I've found a workaround, I simply put a full height black background on the modal holder view when the modal is opened and then it works perfectly.

No rush, but very strange.

```
Android 8 (API26)
RN 0.48
react-native-camera 0.12.0
```

viacheslav-gotsuliak, 2017-12-06:
Having the same issue, 

@l1br3 Could you please explain how to make this black background to fix that transparency? I tried wrapping <Camera> to <View> with black backgroundColor but this doesn't help.

Thanks.

l1br3, 2017-12-06:
@viacheslav-gotsuliak Simply apply a background on the background view, the one which is behind the modal. 
When the modal is open, there is a black overlay background on the view under the modal.

vladimirnani, 2017-12-07:
**opacity: 0.9** did the trick for me otherwise camera couldnt get through

```
<View>
    <Modal>
        <View style={{ flex: 1, backgroundColor: "black", opacity: 0.9999 }}>
            <Camera style={{ flex: 1 }}>
            </Camera>
        </View>
    </Modal>
</View>
```

viacheslav-gotsuliak, 2017-12-07:
There is no such transparency on real devices, so it's not so critical. (Tested on Android as of now, later will test on ios devices)

vladimirnani, 2017-12-07:
@viacheslav-gotsuliak Thanks just tested and confirm no issue on real device.

viacheslav-gotsuliak, 2017-12-12:
Tested on real iPhone, works fine, no transparency.

l1br3, 2017-12-13:
Nice to ear ! I must admit that I didn't even gave it a try on real device !

l1br3, 2017-12-13:
I close the issue as it's non blocking. Feel free to re open it if you think it worth it :)

closed 2017-12-13 by l1br3