react-native-camera/react-native-camera #986: runtime permission for camera in android

milon87, 2017-11-16:
Please add runtime permission to access camera

jgcmarins, 2017-11-16:
Hello, @milon87.
These [permissions](https://github.com/lwansbrough/react-native-camera#permissions) aren't good enough?

milon87, 2017-11-16:
@jgcmarins  yeah but if I use SDK 26 then it shows a white screen instead of the camera. If permission provided manually then it works again as expected.

pavanmehta91, 2017-11-16:
Same issue here. It does not ask for permission and the app crashes due to it but If provided manually by going to the settings it works.

axsann, 2017-11-17:
Use PermissionsAndroid.request() or [react-native-permissions](https://github.com/yonahforst/react-native-permissions).

pavanmehta91, 2017-11-17:
in IOS i don't have to explicitly ask for permission, Do I've to ask for permissions for android explicitly?

milon87, 2017-11-17:
@pavanmehta91  yes

pavanmehta91, 2017-11-17:
Thanks. I implemented the permissions using react-native-permissions module

tungbt-2-3-1994, 2018-06-26:
Hello Mr. Pavanmehta91, 
Could you share your code to implement react-native-permissions? 
I follow sample code in this component, but request function occurs wrong.
Thanks!

closed 2019-06-08 by stale[bot]