react-native-camera/react-native-camera #987: Crash when capture image and pass metadata

koswarabilly, 2017-11-17:
THIS IS SO FRUSTRATING 
METADATA can not be pass it crash everytime i do that
is there any way to fix that?

if this is a duplicate issue please ref me to the solved one

zahid-dapperapps, 2017-12-05:
+1 having same issue, metadata not being saved

zahid-dapperapps, 2017-12-05:
@koswarabilly did you find solution?

koswarabilly, 2017-12-05:
Nope i dont @zahid-dapperapps

zahid313, 2017-12-21:
@koswarabilly  this file https://github.com/npomfret/react-native-camera/blob/c6aede1c092e9d41b4e47b29f80b69f68de0b1c9/ios/RCTCameraManager.m is supposed to be in master but donot why its not, but If you manually replace the existing one with this it should solve your issue
https://github.com/lwansbrough/react-native-camera/pull/635

closed 2019-06-08 by stale[bot]