react-native-camera/react-native-camera #991: Doesn't seem to be picking up qr codes

IljaDaderko, 2017-11-19:
I implemented component to scan qr codes:

```
  // -- methods -- //
  onBarcode = event => {
    AlertIOS.alert("Barcode Found!", "Type: " + event.type + "\nData: " + event.data);
    this.props.camera.setBarcode(event.data);
  };

  // --- render --- //
      <Camera
        onBarCodeRead={this.onBarcode}
        barCodeTypes={["qr"]}
        aspect={Camera.constants.Aspect.fill}
        {...props}
      />
```

Camera is working fine, but I don't seem to be getting `onBarCodeRead` firing when I try it on various barcodes, is there a known issue with this? Or perhaps I didn't understand implementation well?

koswarabilly, 2017-11-20:
@IljaDaderko if your application do not specified just to scan qr code you do not have to declare barcodeTypes

IljaDaderko, 2017-11-20:
@koswarabilly it indeed needs to only work with qr type, I removed ` barCodeTypes={["qr"]}` and it started to work 👍 however I would like to add it back after the fix, to ensure only qr codes are being scanned for.

koswarabilly, 2017-11-20:
@IljaDaderko i think you can use validation instead of declaring barcode type, let the barcodetype as default then make validation in your onBarcode and check event.type , if it is qr then do the action you want if not return null, anyway have you successfully insert location metadata to image capture?

IljaDaderko, 2017-11-21:
@koswarabilly I wasn't able to insert any metadata I'm afraid, as this is not a part of my app. As for qr specific stuff, yeh thats the approach I took, but will leave this open as this feature was documented in docs and appears to be broken

fmsouza, 2018-03-06:
@IljaDaderko I was having the same problem after updating to the latest `react-native-camera` release. Then when I tried to do some logs I realized that the qrcode id changed from `qr` to `org.iso.QRCode`.

What solved the issue for me was using the constants provided by the module itself:

```javascript
import { RNCamera } from 'react-native-camera';

const { BarCodeType } = RNCamera.Constants;

// ...

<RNCamera
  style={styles.camera}
  barCodeTypes={[BarCodeType.qr]}
  onBarCodeRead={this.onBarCodeRead} />

// ...

onBarCodeRead({ type, data }) {
  if (type === BarCodeType.qr) {
    // do something
  }
}
```

closed 2018-03-06 by sibelius