react-native-camera/react-native-camera #992: NullPointerException

dasouch, 2017-11-19:
java.lang.NullPointerException: null
    at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent(RCTCameraViewFinder.java:363)
    at android.view.View.dispatchTouchEvent(View.java:7127)
    at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2176)
    at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1877)
    at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2176)
    at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1877)
    at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2176)
    at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1877)
    at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2176)
    at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1877)
    at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2176)
    at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1877)
    at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2176)
    at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1877)
    at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2176)
    at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1877)
    at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2176)
    at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1877)
    at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2176)
    at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1877)
    at com.android.internal.policy.impl.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:1925)
    at com.android.internal.policy.impl.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1379)
    at android.app.Activity.dispatchTouchEvent(Activity.java:2396)
    at com.android.internal.policy.impl.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:1873)
    at android.view.View.dispatchPointerEvent(View.java:7307)
    at android.view.ViewRootImpl.deliverPointerEvent(ViewRootImpl.java:3172)
    at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:3117)
    at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:4153)
    at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:4132)
    at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:4224)
    at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:171)
    at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
    at android.os.MessageQueue.next(MessageQueue.java:125)
    at android.os.Looper.loop(Looper.java:124)
    at android.app.ActivityThread.main(ActivityThread.java:4745)
    at java.lang.reflect.Method.invokeNative(Method.java)
    at java.lang.reflect.Method.invoke(Method.java:511)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
    at dalvik.system.NativeStart.main(NativeStart.java)

haikyuu, 2017-11-23:
It causes a crash on our app in different android versions.
Here is the code used:
```js
<Camera
  ref={camera => {
    if (camera) this.camera = camera
  }}
  aspect={Camera.constants.Aspect.fill}
  captureMode={Camera.constants.CaptureMode.still}
  captureTarget={Camera.constants.CaptureTarget.temp}
  style={styles.camera}
  type={Camera.constants.Type.front}
>
  <View style={styles.cameraButtonContainer}>
    <TouchableOpacity
      onPress={this.handleTakeSelfie}
      style={styles.cameraButton}
    />
  </View>
</Camera>
```
 Here is the stacktrace:
`
java.lang.NullPointerException:  
at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent (RCTCameraViewFinder.java:349) 
at android.view.View.dispatchTouchEvent (View.java:10779) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.widget.ScrollView.dispatchTouchEvent (ScrollView.java:688) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864) 
at android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491) 
at com.android.internal.policy.DecorView.superDispatchTouchEvent (DecorView.java:605) 
at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent (PhoneWindow.java:1895) 
at android.app.Dialog.dispatchTouchEvent (Dialog.java:894) 
at com.android.internal.policy.DecorView.dispatchTouchEvent (DecorView.java:567) 
at android.view.View.dispatchPointerEvent (View.java:11008) 
at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent (ViewRootImpl.java:5155) 
at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess (ViewRootImpl.java:5007) 
at android.view.ViewRootImpl$InputStage.deliver (ViewRootImpl.java:4532) 
at android.view.ViewRootImpl$InputStage.onDeliverToNext (ViewRootImpl.java:4585) 
at android.view.ViewRootImpl$InputStage.forward (ViewRootImpl.java:4551) 
at android.view.ViewRootImpl$AsyncInputStage.forward (ViewRootImpl.java:4684) 
at android.view.ViewRootImpl$InputStage.apply (ViewRootImpl.java:4559) 
at android.view.ViewRootImpl$AsyncInputStage.apply (ViewRootImpl.java:4741) 
at android.view.ViewRootImpl$InputStage.deliver (ViewRootImpl.java:4532) 
at android.view.ViewRootImpl$InputStage.onDeliverToNext (ViewRootImpl.java:4585) 
at android.view.ViewRootImpl$InputStage.forward (ViewRootImpl.java:4551) 
at android.view.ViewRootImpl$InputStage.apply (ViewRootImpl.java:4559) 
at android.view.ViewRootImpl$InputStage.deliver (ViewRootImpl.java:4532) 
at android.view.ViewRootImpl.deliverInputEvent (ViewRootImpl.java:7092) 
at android.view.ViewRootImpl.doProcessInputEvents (ViewRootImpl.java:7024) 
at android.view.ViewRootImpl.enqueueInputEvent (ViewRootImpl.java:6985) 
at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent (ViewRootImpl.java:7202) 
at android.view.InputEventReceiver.dispatchInputEvent (InputEventReceiver.java:185) 
at android.os.MessageQueue.nativePollOnce (Native Method) 
at android.os.MessageQueue.next (MessageQueue.java:323) 
at android.os.Looper.loop (Looper.java:136) 
at android.app.ActivityThread.main (ActivityThread.java:6776) 
at java.lang.reflect.Method.invoke (Native Method) 
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1518) 
at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1408)
`

The Java code that's causing this issue is the following. It seems `_camera` is `null`, so we'll need to add a check about that.
```java
@Override
    public boolean onTouchEvent(MotionEvent event) {
        // Get the pointer ID
        Camera.Parameters params = _camera.getParameters();
        int action = event.getAction();

        if (event.getPointerCount() > 1) {
            // handle multi-touch events
            if (action == MotionEvent.ACTION_POINTER_DOWN) {
                mFingerSpacing = getFingerSpacing(event);
            } else if (action == MotionEvent.ACTION_MOVE && params.isZoomSupported()) {
                _camera.cancelAutoFocus();
                handleZoom(event, params);
            }
        } else {
            // handle single touch events
            if (action == MotionEvent.ACTION_UP) {
                handleFocus(event, params);
            }
        }
        return true;
    }
```

nes123, 2017-12-06:
we get this error for devices with android 4.*

it will be great to hear some insights on how to prevent that

ospfranco, 2018-01-13:
Please try master version and/or the new RNCamera implementation and see if it fixes your problem, you can read about it in the README

closed 2018-12-19 by n1ru4l