react-native-camera/react-native-camera #995: Fail to mock react-native-camera

assafb81, 2017-11-20:
I am trying to run a basic test in jest:

```
import 'react-native';
import React from 'react';
// Note: test renderer must be required after react-native.
import renderer from 'react-test-renderer';
import Index from '../index.android';

import * as mockCamera from '../__mocks__/react-native-camera';
jest.mock('react-native-camera', () => mockCamera);

it('renders correctly', () => {
    const tree = renderer.create(<Index />);
    expect(tree).toMatchSnapshot();
});
```
**I am getting this error message**:
`Invariant Violation: Element type is invalid: expected a string (for built-in components) or a class/function (for composite components) but got: object. You likely forgot to export your component from the file it's defined in.`
`at invariant (node_modules/fbjs/lib/invariant.js:42:15) at createFiberFromElementType (node_modules/react-test-renderer/cjs/react-test-renderer.development.js:1855:5)
at createFiberFromElement (node_modules/react-test-renderer/cjs/react-test-renderer.development.js:1800:15)
at reconcileSingleElement (node_modules/react-test-renderer/cjs/react-test-renderer.development.js:3172:19)
at reconcileChildFibers (node_modules/react-test-renderer/cjs/react-test-renderer.development.js:3282:37)
at reconcileChildrenAtPriority (node_modules/react-test-renderer/cjs/react-test-renderer.development.js:3925:30)
at reconcileChildren (node_modules/react-test-renderer/cjs/react-test-renderer.development.js:3903:5)
at updateHostRoot (node_modules/react-test-renderer/cjs/react-test-renderer.development.js:4078:7)
at beginWork (node_modules/react-test-renderer/cjs/react-test-renderer.development.js:4434:16)
at performUnitOfWork (node_modules/react-test-renderer/cjs/react-test-renderer.development.js:6030:16)`

I am mocking react-native-camera component in react-native-camera.js file under _ _mocks_ _ folder:
import React from 'react';
```
const constants = {
    Aspect: {},
    BarCodeType: {},
    Type: {},
    CaptureMode: {},
    CaptureTarget: {},
    CaptureQuality: {},
    Orientation: {},
    FlashMode: {},
    TorchMode: {},
};

class Camera extends React.Component {
    static constants = constants;
    render() {
        return null;
    }
}

Camera.constants = constants;
export default Camera;
```

my package.json is:
```
{
    "name": "myApp",
    "version": "0.0.1",
    "private": true,
    "scripts": {
        "start": "node node_modules/react-native/local-cli/cli.js start",
        "lint": "eslint src",
        "test": "jest --runInBand",
        "test:coverage": "jest --coverage --runInBand",
        "build:unsigned:apk": "cd android && ./gradlew clean && ./gradlew assembleRelease -console plain && cd app/build/outputs/apk && ren app-release-unsigned.apk ppg-unsigned.apk"
    },
    "dependencies": {
        "jsrsasign": "^8.0.4",
        "lodash": "^4.17.4",
        "prop-types": "^15.6.0",
        "react": "16.0.0-alpha.12",
        "react-dom": "16.0.0-alpha.12",
        "react-native": "0.48.4",
        "react-native-camera": "^0.10.0",
        "react-native-communications": "^2.2.1",
        "react-native-datepicker": "^1.6.0",
        "react-native-i18n": "^2.0.9",
        "react-native-progress": "^3.4.0",
        "react-native-push-notification": "^3.0.1",
        "react-native-vector-icons": "^4.4.2",
        "react-native-video": "^2.0.0",
        "react-navigation": "^1.0.0-beta.15",
        "react-redux": "^5.0.6",
        "redux": "^3.7.2",
        "redux-thunk": "^2.2.0",
        "reselect": "^3.0.1"
    },
    "devDependencies": {
        "babel-eslint": "8.0.1",
        "babel-jest": "21.2.0",
        "babel-preset-react-native": "4.0.0",
        "enzyme": "^3.1.0",
        "enzyme-adapter-react-16": "^1.0.2",
        "eslint": "4.10.0",
        "eslint-config-airbnb": "16.1.0",
        "eslint-plugin-babel": "4.1.2",
        "eslint-plugin-import": "2.8.0",
        "eslint-plugin-jsx-a11y": "6.0.2",
        "eslint-plugin-react": "7.4.0",
        "jest": "21.2.1",
        "jest-teamcity-reporter": "^0.7.0",
        "react-test-renderer": "16.0.0-alpha.12",
        "redux-mock-store": "^1.3.0"
    },
    "jest": {
        "preset": "react-native",
        "testResultsProcessor": "jest-teamcity-reporter",
        "collectCoverageFrom": [
            "src/**/*.js"
        ],
        "coverageReporters": [
            "html",
            "text"
        ],
        "modulePathIgnorePatterns": [
            "<rootDir>/src/api/",
            "<rootDir>/src/config/",
            "<rootDir>/src/navigators",
            "<rootDir>/src/nls",
            "<rootDir>/src/selectors",
            "<rootDir>/src/store",
            "<rootDir>/src/reducers/index.js"
        ],
        "transformIgnorePatterns": [
            "node_modules/(?!(jest-)?react-native|react-navigation)"
        ],
        "setupFiles": [
            "<rootDir>/test/globalFetch.js",
            "<rootDir>/test/jestSetup.js",
            "<rootDir>/__mocks__/react-native-i18n"
        ]
    }
}
```

hawkup, 2017-12-02:
@assafb81 
I guess because your index.android.js it's not export component
It just call AppRegistry.registryComponent 

**My index.android.js**
```js 
import { AppRegistry } from 'react-native';
import App from './App';
AppRegistry.registerComponent(Appname, () => App);
```

Change to Import ./App on your test file for test

sibelius, 2017-12-09:
you should mock react-native-camera to work with jest

example

__mocks__/react-native-camera.js

```
const Camera = {};
export default Camera;
```

closed 2017-12-09 by sibelius