react-native-camera/react-native-camera #996: $500 bounty to help fix Android issues.

harrisrobin, 2017-11-21:
Hello!

I'm having a bit of an issue with this package on Android. on iOS it runs supper smoothly 👌

On Android, the performance is pretty bad and I keep getting the following 2 crashes:

## Crash 1
```
Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference
```

## Crash 2
```
Unable to resume activity {com.heapr/com.heapr.MainActivity}: java.lang.ArrayIndexOutOfBoundsException: length=0; index=0
```

I will pay $500 to anyone who can either assist me in fixing these issues or who manage to land a PR that fixes them on Android. 
Bonus: if someone can get Android performance to improve as well, I will throw in a bonus.

Thanks,

charpeni, 2017-11-21:
You can ping me @harrisrobin for the review & merge.

harrisrobin, 2017-11-21:
@grabbou Yes absolutely! I'll get it done today and get back to you here with confirmation.

I just have 2 questions:

1. When do you think would be reasonable for me to expect something I can test?
2. Is there anything I can do to help you and/or make your life easier with this?

Thanks!

grabbou, 2017-11-21:
@harrisrobin - no need to do it upfront. I think it's better to wait till I ship PR (which can happen quickly depending on the issue).

I'll dig into this today and get back to you re points 1 and 2.

RZulfikri, 2017-11-21:
Can you tell us more about your issue, like step to step when you got that issue and your android device.

harrisrobin, 2017-11-21:
@RZulfikri I will get back to you tomorrow with more details and documented steps.

grabbou, 2017-11-23:
Update from my side: I will be able to look into that on Monday. If anyone wants to start earlier, please do. Otherwise, I'll be looking into it in few days.

harrisrobin, 2017-12-13:
Update :
This is still on!
I have gotten a lot of feedback which has helped but I am still offering the $500 bounty to anyone that can land a PR that improves this library and it's stability, especially when it comes to crashes. I have not had enough time to show examples of the issues i am running into on Android but hoping to get that done by end of week :)

Thanks for following up everyone!

amscher, 2017-12-14:
@harrisrobin I'm having the same issues, however I'm a noob to react native and react navigation, so I'm curious how your "smart rendering of the camera" works as well as what you're using for crash reporting.

CrazyPython, 2018-01-05:
you can post the bounty on http://bountysource.com, with a deposit

sibelius, 2018-01-05:
@harrisrobin check if this PR solves your problem

https://github.com/react-native-community/react-native-camera/pull/1071

run the example provided, and try in your production app

this new camera implementation is only for android for now, and the api is the same as expo camera module

afilp, 2018-01-07:
@sibelius This has 84 files changed, when do you think it can be merged? We really need this soon because Android apps crash always after some scans. Thanks again!

CrazyPython, 2018-01-11:
@afilp https://stackoverflow.com/a/36743127/1459669

sibelius, 2018-01-23:
RNCamera is on master for both android and ios

https://github.com/react-native-community/react-native-camera#experimental

closed 2018-01-23 by sibelius