react-native-camera/react-native-camera #997: Tried to register two views with the same name RCTCamera

PsArvin, 2017-11-21:
Device: Android 7.0
react-native: 0.50.3
react-native-camera: 0.12.0

[image]

krizpoon, 2017-11-29:
I met the same problem. In my case, there is another library that declares `react-native-camera` as dependency. I fixed it by deleting the `node_modules/react-native-camera` folder inside that library.

sibelius, 2018-02-12:
is this an issue?

kuailetuoluoren, 2018-06-05:
i meet the same problem,
    "react": "16.3.1",
    "react-native": "0.55.4",
    "react-native-camera": "^1.1.2",

ducanhmai205, 2018-07-09:
how can i fix this issue

JulianKingman, 2018-07-21:
@krizpoon my hero. I found a few instances of the package in question (in my case `react-native-svg`) installed in other dependencies. Removing those solved the problem. I'd imaging that version mismatches between them contribute to the issue, as I was attempting to change the version number..

closed 2019-04-18 by sibelius