react-native-camera/react-native-camera #999: Destroy camera between screens

oricoh, 2017-11-21:
Hi,

I have encountered two problems when using the camera in different screens in the same app.

1. In my app, I load the camera in forced Portrait screen, and later in a different screen I load it in a forced landscape. However the first screen sets the size of the camera and in the second screen it comes up squished and narrow in the center of the screen with the width of the first camera.
2. In another case, I load the camera with a different set of GUI/buttons in 2 different screens, but the result is that in the second screen I see both the GUI one on the top of each other.

How can I stop/destroy the camera so when I load it in a different screen, it comes up as new?

Thanks!

harrisrobin, 2017-11-21:
If you are using react-navigation, you need to dispatch a reset action so that the camera is unmounted. 

See https://github.com/lwansbrough/react-native-camera/issues/701 for more details.

jslok, 2017-11-22:
I showed an example of how I unmount the camera between screens using redux here #878

diegoool, 2019-06-27:
Ive used withNavigationFocus with HOC and this helped me!

import { withNavigationFocus } from 'react-navigation';

class MyWrapComponent extends React.Component {
render() {
return this.props.isFocused ? < CameraComponent /> : <></>;
}
}

export default withNavigationFocus(MyWrapComponent );

michael-harley, 2019-11-07:
> Ive used withNavigationFocus with HOC and this helped me!
> 
> import { withNavigationFocus } from 'react-navigation';
> 
> class MyWrapComponent extends React.Component {
> render() {
> return this.props.isFocused ? < CameraComponent /> : <></>;
> }
> }
> 
> export default withNavigationFocus(MyWrapComponent );

@diegoool This doesn't work! :(
What version of react, react-native, react-native-camera, react-navigation did you use?
You have another solution maybe? I already tried the Navigation Stack reset too. Still get stuck on react native camera (Happening sometimes, but quite a lot)

iDevo, 2020-06-09:
Hi @michael-harley! Did you find a solution? I am updated the code to the new react navigation 5 and also tried to use old APIs but unfortunately without success. Switching the camera from one screen to the other leads to a black RNCamera screen.

closed 2020-08-15 by stale[bot]

CorentinHeroux, 2021-02-23:
Same issue.