react-native-camera/react-native-camera #1000: Takes a lot of time to take/show and store image

bvv218, 2017-11-22:
It takes 3-4 sec to show photo after taking it and 6-8 sec to store it. I am using the example and it's the same. Is there a way to increase the speed?

stianjensen, 2017-11-22:
Is this on android? In that case this is a duplicate of #768

bvv218, 2017-11-23:
@stianjensen Yes, it's on android, not tested on ios. #768 says that delay is only to store image (I have had this before as well), but now I started to have the delay between taking the picture and showing the preview too. Is there a way to fix at least this part of the problem?

stianjensen, 2017-11-23:
Not that I’m aware of. I have the same problem wherre even the preview is slow to show up.

KS-Balaji, 2017-11-23:
I have exactly the same problem... any solutions...?

ducpt2, 2017-12-01:
+ 1. same problem...

MehdiAlouafi, 2017-12-05:
+ 1

AlexandreMaistret, 2018-01-09:
+1 Did someone have find any solution ?  Because it is really too slow and I have tried on two smartphones Wiko and Samsung, and the two are really really slow when a photo is taken.

ospfranco, 2018-04-03:
added skipProcessing flag (android only) for the takePictureAsync call params, this should make picture snapping super fast, but might break rotation on some of your apps.

Please give it a go on the master branch.

closed 2018-04-03 by sibelius

andreh111, 2018-05-05:
hey guys try using this https://www.npmjs.com/package/react-native-fast-image 
it can solve the problem of image loading time

lucasluca, 2020-02-06:
i was using this config:
const options = { quality: 0.1, base64: false, forceUpOrientation: true, fixOrientation: true, skipProcessing: true };

and then i removed {forceUpOrientation: true, fixOrientation: true}, from this object and got it working really fast

devartwa, 2020-12-04:
i removed fixOrientation: false and worked for me.