react-native-camera/react-native-camera #1001: PropTypes error in RN 0.50.x

moonjava2005, 2017-11-23:
Dear all,
I try using react-native-camera version 1.0.. alpha
I found incompatible error with RN 0.50.x
Should change
import React, {Component} from 'react';
to
import PropTypes from 'prop-types';
Best regards,

thatjuan, 2017-11-28:
Are you sure you're pulling the right package? Current version of this package is 0.12.0 and it has no PropType errors on RN 0.50

moonjava2005, 2017-11-29:
Dear you guys,
I'm using 'react-native-camera' version 1.0.0-alpha1

hawkup, 2017-12-02:
@moonjava2005 the latest version is 0.12.0 on **Master** branch
1.0.0-alpha1 is the version on [**Next** branch](https://github.com/lwansbrough/react-native-camera/blob/ddd9ba7c9fa5e6146312b172bd3d97e3539da028/package.json#L7) but it has not updated since Jan 2017

ospfranco, 2018-01-13:
rn ~0.50.0 support should be on master, read the README and update accordingly

closed 2018-01-13 by ospfranco