react-native-camera/react-native-camera #1002: Barcode/QR scanning too slow.

zurez, 2017-11-24:
Camera is very slow to identify the barcode & QR.
Any way to improve performance?

IljaDaderko, 2017-11-26:
what rn and react-native-camera versions are you running? It scans barcode blazing fast on my phone, even without focusing on it long.

```
      onBarcode = event => {
         console.log(event);
      }; 

      <Camera
        onBarCodeRead={this.onBarcode}
        aspect={Camera.constants.Aspect.fill}
      />
```

IljaDaderko, 2017-11-26:
Try upgrading to these

```
"react-native": "0.50.4",
"react-native-camera": "0.12.0",
```

Georgevik, 2017-12-01:
I read too slow too when I'm reading from a display

inev, 2017-12-01:
Are you on android? From what I can gather it's a lot slower on android due to this issue: https://github.com/lwansbrough/react-native-camera/issues/945

Georgevik, 2017-12-04:
Thank you @inev, but finally I have used the recommendations from the issue #664 (ZBar library) for Android and it's go like a charm.And for iOS, I am using this library that is really good

zurez, 2017-12-04:
@Georgevik  How did you use ZBar? I am new to RN and therefore any guide would be helpful

Georgevik, 2017-12-04:
Sorry, I was confused. I have just checked the code and I didn't use ZBar. The final library for Android is https://github.com/ekreutz/react-native-barcode-scanner-google that works really good only for Android

martwetzels, 2017-12-21:
I'm experiencing the same issue on RN (0.50.4) and react-native-camera (0.12.0). For Android, I noticed that holding the camera 'still' in front of the QR code will take very long to detect (or not even detect). However, when slighting moving/shaking the phone will result in an almost instant detection. 

Is there some internal mechanism in the Android implementation that needs a certain amount of pixels to be changed between frames before the detection is run?

abdulrahman-khankan, 2017-12-28:
Same thing on Android, trying to read EAN-13 codes but it's too slow. With Expo's API demo app it's super fast and it's built on top of this :/ Not sure if I'm doing something wrong. 
RN: 0.50.3
RNCamera: 0.12

sibelius, 2017-12-28:
expo uses another camera implementation

abdulrahman-khankan, 2017-12-28:
@sibelius thanks for clarifying that! 
Any suggestions on how to improve the reader for EAN-13 on Android?

sibelius, 2017-12-28:
I think the idea is to move android code closer to what expo does https://github.com/expo/expo/tree/master/android/expoview/src/main/java/versioned/host/exp/exponent/modules/api/components/camera

using cameraview

it would be a major overhaul, but it would solve a lot of issues on android

JorgeCeja, 2018-03-01:
Any updates on your previous post? Thanks! @sibelius

sibelius, 2018-03-01:
It should be fixed with RNCamera on 1.0.0

ShahoG, 2018-03-28:
I'm sorry but this issue still persists on android

karnash, 2018-04-30:
show too slow when the camera show

RyanRouton-Rival, 2018-05-25:
@sibelius Has anyone been able to profile this?  As soon as you enable barcode reading it drops frames down to about 4 a second on a Nexus 6p.

sibelius, 2018-05-25:
you can try to use google visio bar code scanning on android

it should work well

kigawas, 2018-09-21:
Hey guys, I found how to use google vision API with react-native-camera.

You can set `onGoogleVisionBarcodesDetected` like this:
```js
<RNCamera
    ref={ref => {
        this.camera = ref
    }}
    style={YOUR_STYLE_TO_SHOW_CAMERA_VIEW}
    type={RNCamera.Constants.Type.back}
    barCodeTypes={[RNCamera.Constants.BarCodeType.qr]}
    onGoogleVisionBarcodesDetected={
        (e) => {
            console.warn(e)
        }
    }
/>
```

Ref: https://github.com/react-native-community/react-native-camera/issues/542#issuecomment-423419379

closed 2019-06-08 by stale[bot]

MohsenKhosravinia, 2019-12-16:
If you are using AVFoundation, you should use below metadata object types:

               metadataOutput.metadataObjectTypes = [
                AVMetadataObject.ObjectType.aztec,
                AVMetadataObject.ObjectType.code128,
                AVMetadataObject.ObjectType.code39,
                AVMetadataObject.ObjectType.code39Mod43,
                AVMetadataObject.ObjectType.code93,
                AVMetadataObject.ObjectType.dataMatrix,
                AVMetadataObject.ObjectType.ean13,
                AVMetadataObject.ObjectType.ean8,
                AVMetadataObject.ObjectType.face,
                AVMetadataObject.ObjectType.interleaved2of5,
                AVMetadataObject.ObjectType.itf14,
                AVMetadataObject.ObjectType.pdf417,
                AVMetadataObject.ObjectType.qr,
                AVMetadataObject.ObjectType.upce]

and you can inform users that they should not bring the camera too close to the barcode. (for EAN barcodes)

HyopeR, 2022-05-19:
I had this problem in a project I was developing. Barcode and Qr code readings were working really slow on the android platform. I will share the steps I took to fix the problem with code examples.

First of all, we will use the **different callback functions** provided by the **RNCamera** component on the two platforms.

Let's create **constants** so that we can use them on both platforms.

<pre>
import {RNCamera} from 'react-native-camera';
import {Platform} from 'react-native';
const DefaultBarcodeType = RNCamera.Constants.BarCodeType;
const GoogleBarcodeType =
  RNCamera.Constants.GoogleVisionBarcodeDetection.BarcodeType;

// Barcodes by platform
export const BARCODES = Platform.select({
  ios: {
    AZTEC: DefaultBarcodeType.aztec,
    CODE_128: DefaultBarcodeType.code128,
    CODE_39: DefaultBarcodeType.code39,
    EAN_13: DefaultBarcodeType.ean13,
    EAN_8: DefaultBarcodeType.ean8,
    PDF417: DefaultBarcodeType.pdf417,
    QR_CODE: DefaultBarcodeType.qr,
    UPC_E: DefaultBarcodeType.upc_e,
    ALL: null,
  },
  android: {
    AZTEC: GoogleBarcodeType.AZTEC,
    CODE_128: GoogleBarcodeType.CODE_128,
    CODE_39: GoogleBarcodeType.CODE_39,
    EAN_13: GoogleBarcodeType.EAN_13,
    EAN_8: GoogleBarcodeType.EAN_8,
    PDF417: GoogleBarcodeType.PDF417,
    QR_CODE: GoogleBarcodeType.QR_CODE,
    UPC_E: GoogleBarcodeType.UPC_E,
    ALL: GoogleBarcodeType.ALL,
  },
});

// Helper function
// Anrdroid side Google constants express the sum of number values. 
// For example: [CODE_128, QR_CODE] = 1 + 156 = 157.
// The camera will only read these two code types.

export const getBarcodeTypes = (barcodeTypes) => {
  return Platform.select({
    ios: barcodeTypes
    android: barcodeTypes.reduce((prev, current) => prev + current, 0)
  })
}
</pre>

**For ios platform usage RNCamera:**
```
<RNCamera
      barCodeTypes={getBarcodeTypes([BARCODES.CODE_128, BARCODES.QR_CODE])}
      onBarCodeRead={() => {}}
/>
```

**For android platform usage RNCamera:**
```
<RNCamera
      googleVisionBarcodeType={getBarcodeTypes([BARCODES.CODE_128, BARCODES.QR_CODE])}
      onGoogleVisionBarcodesDetected={() => {}}
/>
```
In my experience, **onGoogleVisionBarcodesDetected works much faster on android platform**. When you set up a structure in this way, you can specify specific barcode scans on both platforms and get fast results.