react-native-camera/react-native-camera #1004: Problem to install react-native-camera  platform tools 2.3

devSeb, 2017-11-25:
Hello everybody,

I would like to intall react-native-camera 
I follow the breaking changes , if i want to use run-android,  I have a prob with buildToolsVersion.
I have  \Android\sdk\build-tools\23.0.1
I use Android studio to install api and build tools, but if i would like to install 25.2 or 26 or 27, I have the same problem.

err : 
Warning: License for package Android SDK Platform 23 not accepted.

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':app'.
> You have not accepted the license agreements of the following SDK components:
[Android SDK Platform 23, Android SDK Build-Tools 23.0.1].
Before building your project, you need to accept the license agreements and complete the installation of the mi                                                                         ssing components using the Android Studio SDK Manager.
Alternatively, to learn how to transfer the license agreements from one workstation to another, go to http://d.                                                                         android.com/r/studio-ui/export-licenses.html

How to resolve this ?

closed 2017-11-25 by devSeb

hcandia-lagash, 2017-12-05:
@devSeb Did you resolve it?

devSeb, 2017-12-05:
hello, 
Yes, It's an error of me. I clean the graddle and it's working