react-native-camera/react-native-camera #1005: undefined is not an object ( evaluating 'react2.propTypes.oneOfTypes')

devSeb, 2017-11-25:
Hello everybody,
I would like to implement react-native-camera in my project.
If I run the example in my project , I have an error 
" undefined is not an object ( evaluating 'react2.propTypes.oneOfTypes') "
<unknown> .../node_modules/react-native-camera/index.js:75:22

my package.json

`	"dependencies": {
		"react": "^15.4.0",
		"react-dom": "^15.4.0",
		"react-native": "0.50.4",
		"react-native-camera": "^0.6.0"
	},
	"devDependencies": {
		"babel-jest": "21.2.0",
		"babel-preset-react-native": "4.0.0",
		"jest": "21.2.1",
		"react-test-renderer": "16.0.0"
	},
	"jest": {
		"preset": "react-native"
	}`

My app.js is the same example of the readme

pengyongqi, 2017-11-28:
我也遇到了这个问题，我是这样解决的，希望能帮到你：

1.使用  rnpm  link，如果rnpm没有安装，先  yarn add rnpm  或者 npm install rnpm --save

2.在MainApplication.java中加入

//摄像头
import com.lwansbrough.RCTCamera.RCTCameraPackage;

getPackages（）中加入

new RCTCameraPackage()

试试看行不行。

my English is pool,I hope it's work for you！

devSeb, 2017-11-28:
hello, thanks for you message.
I do this in my project.
rnpm link react-native-camera,   I don't understand why i have a problem with a props in the index.js in the library with the example in the readme

hawkup, 2017-11-29:
If you use react-native-camera version 0.6.0, PropTypes import from React package
```js
// react-native-camera/index.js
import React, { Component, PropTypes } from 'react';
```
And if you use React version 15.5.0 or higher,  PropTypes move to own package (reference: https://reactjs.org/blog/2017/04/07/react-v15.5.0.html), have to change import follow this
```js
// react-native-camera/index.js
import React, { Component } from 'react';
import PropTypes from 'prop-types';
```

Update to the latest version (It's fixed on version 0.10.0)

devSeb, 2017-12-05:
Thanks @hawkup .
It's worked.
This fix must be in the source code I think.

Is there a pull request for this fix ?

sibelius, 2017-12-31:
fixed on master

closed 2017-12-31 by sibelius

youngjuning, 2018-06-25:
[image]

jsastriques, 2019-03-21:
i`m getting the same error, i´ve tried you suggestions and i don´t get result, here is my error:

 undefined is not an object (evaluating '_react.PropTypes.oneOfType')
Unhandled JS Exception: undefined is not an object (evaluating '_react.PropTypes.oneOfType')