react-native-camera/react-native-camera #1006: SIGABRT when recording video on iOS

thatjuan, 2017-11-28:
**Env:**
React Native: 0.49.3
Yarn: 1.3.2
RCTCamera: 0.12.0

**Issue:**
My Camera declaration:

```
            <Camera
              ref={(cam) => {
                  this.camera = cam;
              }}
              style={{flex:1}}
              aspect={Camera.constants.Aspect.fill}
              captureMode={Camera.constants.CaptureMode.still}
              captureTarget={Camera.constants.CaptureTarget.disk}
              captureQuality={Camera.constants.CaptureQuality.photo}
              type={this.state.cameraType}
              flashMode={this.state.flashMode}
              keepAwake={true}
              onZoomChanged={() => {}}
              onFocusChanged={() => {}}
            />
```

When I call:

```
            this.camera.capture({mode: Camera.constants.CaptureMode.video})
                .then((data) => { ...
```

I get a `SIGABRT` (NSException) crash right on [this line](https://github.com/lwansbrough/react-native-camera/blob/59a0b44c71c7b13601dfd34b43714bf8cbb198eb/ios/RCTCameraManager.m#L786).

I've been trying to debug it, but there's a lot of context to go through. It doesn't seem to be a generic issue that many people run into with AVCaptureMovieFileOutput. There aren't any obvious nils that I can see before the line is reached.

Any clues? Thanks in advance.

thatjuan, 2017-11-28:
I found the crash is caused by specifying `Camera.constants.CaptureQuality.photo` as the captureQuality.

It'd probably make sense to add guards against that.

haofeng123, 2018-01-03:
I think I also had a problem How do you solve that ,thank you

georgewwindsor, 2018-03-06:
I am additionally experiencing this, but only randomly.

sibelius, 2018-03-06:
Upgrade to RNCamera

closed 2018-03-06 by sibelius