react-native-camera/react-native-camera #1008: capture not working in android 6.0.1 (OnePlus)

svenvanlierde, 2017-11-28:
Camera.capture is not working on a OnePlus One android 6.0.1. 
It does not log anything in then or catch...
But in the logcat output i have multiple time (before capture)
af_process_parse_stats: Invalid FV data!
af_process: Error processing AF stats!

react-native: 0.48.3
react-native-camera: 0.12.0

Any suggestions? thanks :)

karltaylor, 2017-12-15:
Also have this problem

karltaylor, 2017-12-19:
@svenvanlierde can you show me your component? I had a problem where I was underendering the camera before it could complete.

svenvanlierde, 2017-12-19:
Hi, 

Changing captureTarget={"temp"} to captureTarget={"disk"} solved my problem on OnePlus...

sibelius, 2018-01-12:
try the experimental RNCamera https://github.com/react-native-community/react-native-camera#experimental

closed 2018-09-24 by SimonErm