react-native-camera/react-native-camera #1017: Cannot read property 'Aspect' of undefined

avalond, 2017-12-02:
in ios i use react-native-camera have this error 'Cannot read property 'Aspect' of undefined'

TheBnl, 2017-12-08:
I have the same issue, the issue is triggered when the native module gets required. 
When simulating on iOS 10 the camera works just fine. 
Maybe the iOS camera API changed for iOS 11?

avalond, 2017-12-08:
@TheBnl  in my all devices is not working, ios 9.3.3 ios 11.2,ios 11.1.3

jonathanleahy, 2017-12-08:
Broken when I publish to android as well.

sibelius, 2017-12-09:
use

`react-native link react-native-camera` and build again

closed 2017-12-09 by sibelius

avalond, 2017-12-10:
@sibelius  not working, i try rm -rf node_model/   reinstall  and relink react-native-camera is not working in ios, please check

sibelius, 2017-12-10:
Can you log all your NativeModules?

And check if there is a RCTCamera on it?

If there is no RCTCamera the link is wrong

Sometimes React-Native link does not work well

jonathanleahy, 2017-12-10:
I tried with publishing to three phones. 2 more modern phones worked fine, the other came up with that error. Same project for all. The one which didn’t work was running android 5.1

avalond, 2017-12-10:
this is my error page 
[image]

this is my js code import react native camera
[image]

this is my xcode link RCTCamera is working 
[image]

but is not working @sibelius

sibelius, 2017-12-10:
Add this snippet to your code

```jsx
import { NativeModules } from 'react-native'

console.log('NativeModules: ', NativeModules);
```

you should have one of this 2

NativeModules.CameraManager || NativeModules.CameraModule;

a CameraManager for ios and a CameraModule for android

avalond, 2017-12-10:
have this messages ExceptionsManager.js:73 Unhandled JS Exception: Cannot read property 'Aspect' of undefined
reactConsoleErrorHandler @ ExceptionsManager.js:73
logToConsole @ RCTLog.js:48
logIfNoNativeHook @ RCTLog.js:31
__callFunction @ MessageQueue.js:302
(anonymous) @ MessageQueue.js:116
__guard @ MessageQueue.js:265
callFunctionReturnFlushedQueue @ MessageQueue.js:115
(anonymous) @ debuggerWorker.js:72
ExceptionsManager.js:73 Unhandled JS Exception: Module AppRegistry is not a registered callable module (calling runApplication)

@sibelius

sibelius, 2017-12-10:
add this snippet before everything on your App.js

you can stop importing react-native-camera to help your debug

avalond, 2017-12-10:
[image]
i got log bug is not have Aspect

sibelius, 2017-12-10:
you do not have Camera modules on your NativeModules

it is not linked properly

try in a clean project

st4allen, 2018-01-03:
I also have this issue. Then I find that I missed two steps in the documentation.

First remember to add permissions in plist file.

```
<key>NSPhotoLibraryAddUsageDescription</key>
<string>Your message to user when the camera is accessed for the first time</string>
<key>NSCameraUsageDescription</key>
<string>Your message to user when the camera is accessed for the first time</string>
<key>NSPhotoLibraryUsageDescription</key>
<string>Your message to user when the photo library is accessed for the first time</string>
<key>NSMicrophoneUsageDescription</key>
```
Then remember to add dependency in your podfile.

`pod 'react-native-camera', :path => '../node_modules/react-native-camera'`
and run `pod install`.

For me it works after these two steps. If it still not work, check if you link the library properly.

Hope it helps.

mashwinprdxn, 2018-06-21:
In case anyone's facing this issue on Android, please try the [gradle upgrade guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/GradleUpgradeGuide.md). It worked for me.

sadeghhosseini, 2018-09-09:
I am facing the same issue, would you please provide a react-native-camera demo app so that beginners can build their project on top of it?