react-native-camera/react-native-camera #1018: Error retrieving camcorder profile params

sm2017, 2017-12-02:
In the following line
https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCamera.java#L266

If we set `captureQuality` as `RCT_CAMERA_CAPTURE_QUALITY_MEDIUM`  and device camera don't support `CamcorderProfile.QUALITY_720P` , App will crash at

```
java.lang.RuntimeException: Error retrieving camcorder profile params
	 at android.media.CamcorderProfile.native_get_camcorder_profile(Native Method)
	 at android.media.CamcorderProfile.get(CamcorderProfile.java:384)
	 at com.lwansbrough.RCTCamera.RCTCamera.setCaptureVideoQuality(RCTCamera.java:266)
	 at com.lwansbrough.RCTCamera.RCTCameraModule.prepareMediaRecorder(RCTCameraModule.java:264)
	 at com.lwansbrough.RCTCamera.RCTCameraModule.record(RCTCameraModule.java:350)
	 at com.lwansbrough.RCTCamera.RCTCameraModule.captureWithOrientation(RCTCameraModule.java:514)
	 at com.lwansbrough.RCTCamera.RCTCameraModule.access$100(RCTCameraModule.java:38)
	 at com.lwansbrough.RCTCamera.RCTCameraModule$2.orientationEvent(RCTCameraModule.java:498)
	 at com.lwansbrough.RCTCamera.RCTSensorOrientationChecker$Listener.onSensorChanged(RCTSensorOrientationChecker.java:64)
	 at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:405)
```

We have issue with `RCT_CAMERA_CAPTURE_QUALITY_LOW` too

Just a suggest , We can use the following code
```java
if(CamcorderProfile.hasProfile(_cameraTypeToIndex.get(cameraType), CamcorderProfile.QUALITY_720P)){
	cm = CamcorderProfile.get(_cameraTypeToIndex.get(cameraType), CamcorderProfile.QUALITY_720P);
}
```

sm2017, 2017-12-09:
Hello ... ??

sibelius, 2017-12-09:
@sm2017 send a pr to fix this please

Polidoro, 2018-02-17:
Getting the same error on the simulator. Running an S7 with 6.0.0

sibelius, 2018-02-18:
Try to use RNCamera

Polidoro, 2018-02-18:
Thanks for the quick reply! Gave it a shot and it looks like I'll need to update Gradle to take advantage of RNCamera, which is a project for another day.

Really appreciate it though, I'll report back once I'm done updating!

SimonErm, 2018-10-11:
https://github.com/react-native-community/react-native-camera/pull/1835

closed 2018-10-11 by SimonErm