react-native-camera/react-native-camera #1021: App Crash on Android

vanhoutk, 2017-12-04:
### Expected behaviour
I've been using this package for the last three months in my app and it's been working perfectly. Opening a view with a `<camera>` component in it accesses the camera, recording works fine and I'm able to do what I want with the recording afterwards.

### Actual behaviour
In the last few days this behaviour has changed and I can't figure out why. When I first launch my app after clearing all data, I can navigate into my view containing a `<camera>` component. When I start to try record, I get a standard android message that the app has crashed. There's no JS logging or Logcat logging appearing for this that I can see.

When I open my app again, I'm unable to navigate into the view containing the `<camera>` component. The render method of the view gets called, but nothing changes on screen unless I remove the `<camera>` component. When I try to navigate to the view the app just hangs at that point until I get a standard android message saying it's not responding and asking if I'd like to close it. Again, no logs that I can see.

If anyone has any thoughts it'd be much appreciated.

### Environment
- **React Native version**: 0.47.2
- **React Native platform + platform version**: Android 6.0.1

### react-native-camera
**Version**: 0.12.0

AleixSanfe, 2017-12-05:
I have the same problem and I'm on my way to try to find a temporal solution.

The problem is caused because android, on the newer versions (+5.0 (API lvl 21) ) has changed all classes about camera as it is explained here: [Android Camera API](https://developer.android.com/guide/topics/media/camera.html)

vanhoutk, 2017-12-05:
After testing on a couple of emulators, it seems to be any API before 25 that it isn't working with.

vanhoutk, 2017-12-05:
Managed to solve this by uninstalling the app, reinstalling, and manually setting the permissions as described at the end of #696 

Strange that it was working fine originally but isn't now.

closed 2017-12-05 by vanhoutk