react-native-camera/react-native-camera #1025: No audio in video / recorded video is broken when audio:true is set

facuescobar, 2017-12-06:
Hi!,

I'm trying to configure video settings, and I get this results:
- Video is recorded, but it has no audio track:
```
this.camera.capture({
              mode: DeviceCamera.constants.CaptureMode.video,
            })
            .then(this.onCaptureSuccess)
            .catch(this.onCaptureError);
```

- Video file gets saved in camera roll, but when I try to play it in camera roll, seems to be broken and stay frozen:
```
this.camera.capture({
              mode: DeviceCamera.constants.CaptureMode.video,
              audio: true
            })
            .then(this.onCaptureSuccess)
            .catch(this.onCaptureError);
```

Camera - audio/captureAudio settings does not seem to have any effect when are added in Camera element:
```
<Camera
          ref={cam => {
            this.camera = cam;
          }}
          style={Style.deviceCamera}
          // Camera Settings
          type={this.state.type}
          captureMode={this.state.captureMode}
          flashMode={this.state.flashMode}
          // iOS Settings
          audio={true}
          // Android Settings
          playSoundOnCapture={true}
>
...
</Camera>
```

Project information
```
Environment:
  OS:  macOS Sierra 10.12.6
  Node:  7.9.0
  Yarn:  0.23.2
  npm:  5.5.1
  Watchman:  4.7.0
  Xcode:  Xcode 8.2.1 Build version 8C1002
  Android Studio:  2.3 AI-162.4069837

Packages: (wanted => installed)
  react: 16.0.0-beta.5 => 16.0.0-beta.5
  react-native: 0.49.3 => 0.49.3
```

Device Information:
```
iPhone 7 
iOS 10.2.1
```

Any thought?

jtokoph, 2018-01-08:
There is an issue with documentation introduced with this: https://github.com/react-native-community/react-native-camera/pull/809

The prop on the Camera component should be `captureAudio` instead of `audio`

jgcmarins, 2018-01-09:
@jtokoph could you help us on the reverting process?

closed 2019-06-08 by stale[bot]