react-native-camera/react-native-camera #1029: rewrite using Swift?

sibelius, 2017-12-08:
What do you guys think in a refactor to rewrite this package in Swift?

sibelius, 2018-01-22:
We can convert both to swift

tirrorex, 2018-01-22:
Dont't know how that will affect the community though, i assume most contributors might be proficient in objective-c but not necessarly in swift?

sibelius, 2018-01-22:
I think swift would make it easier to maintain this package

msand, 2018-03-10:
It seems to be possible to write the iOS parts using Kotlin as well: http://www.albertgao.xyz/2018/01/14/how-to-create-kotlin-native-ios-project/
Wonder if it could lead to more code reuse and even less maintenance burden.

n1ru4l, 2018-03-10:
I am not that deep into android development and iOS development, but since we are using platform specific apis I do not think that writing the code in kotlin would result in that more reusable code.

msand, 2018-03-11:
Yeah, I haven't read the source of this project, so can very well be. At least in react-native-svg I think there is some opportunity for reuse, but it might just overly complicate things as well.

tirrorex, 2018-03-11:
nobody is gonna write ios code in kotlin guys

msand, 2018-03-11:
Sounds a bit like when people were saying noone would write native apps for android and ios using javascript not too long ago 😉 A bit early to judge how it's gonna go, as support for calling kotlin code from obj-c and swift is what, 3 months old? I'm already seeing it happen, and wouldn't be surprised if it keeps growing quite a bit still. As kotlin also compiles to javascript, it seems ripe with potential, and seems to be one of the fastest, if not the fastest growing in popularity. Even Tiobe is likely to chose either Kotlin or C as the language of the year. You have any technical reasons for your statement? Or should it be taken as a challenge? 😄 

http://pypl.github.io/PYPL.html
https://www.tiobe.com/tiobe-index/
https://stackify.com/popular-programming-languages-2018/

msand, 2018-03-11:
The KotlinConf Spinner app looks really interesting:

>     - a backend, using SQLite for data storage and exposing a REST/JSON API;
>     - mobile clients for iOS and Android, using OpenGL;
>     - a WebAssembly-based browser frontend for viewing the game scores.

https://kotlinlang.org/docs/reference/native-overview.html
https://github.com/jetbrains/kotlinconf-spinner

tirrorex, 2018-03-11:
What i meant is since we use javascript already AND native code, there is no need using another language.
Sure i could write something to build ios code in kobol, doesn't mean it's a good idea.
Kotlin official support for android is great but there won't be an official support for kotlin from apple, period.
You think they invested millions in swift just to drop it and support kotlin? never gonna happen ;)

msand, 2018-03-11:
I'm not sure why or how apple would need to be involved in it in any way, can you elaborate? Both Swift and Kotlin use LLVM to do their thing in this case. Headers for any relevant interfaces are available. What more do you need? If there is to be a rewrite, and the Android part would be written in Kotlin, and the same code (or at least the common parts) can be compiled for iOS, what use is there in writing the same thing in Swift as well? Just seems like at least double the effort for very marginal benefit. Of course, writing it in c/c++ might make sense using the same logic, but then DX comes into the picture. Then its more a question of prioritizing either ergonomics of the language or getting rid of the overhead from reference counting and garbage collection. At least, as Kotlin works for both iOS and Android, this could remove at least one or possibly two of the three languages needed. The benefits from that seem obvious to me, perhaps I'm missing something?

fabriziobertoglio1987, 2020-04-06:
We already have a lot of manteinance work..

MarcoScabbiolo, 2021-01-11:
Moving to Kotlin/Swift is the logical thing to do, is what Apple and Google want for their ecosystems.
They are both objectively superior to their legacy counterparts and have a better development experience on every aspect.

> No. React-Native and React-Native-Camera and many other libraries are known to not be ready for production for many reasons:

React-native is production ready,  it is not if you are trying to have native-grade features and performance writing only JS when using SDKs like ML and Camera.

MateusAndrade, 2021-01-11:
> Moving to Kotlin/Swift is the logical thing to do, is what Apple and Google want for their ecosystems.
> They are both objectively superior to their legacy counterparts and have a better development experience on every aspect.
> 
> > No. React-Native and React-Native-Camera and many other libraries are known to not be ready for production for many reasons:
> 
> React-native is production ready, it is not if you are trying to have native-grade features and performance writing only JS when using advanced SDKs like ML and Camera. That is what makes libraries like this one so important

I think we can start moving small chunks of code to both swift/kotlin. How do you think we can start doing this?

MarcoScabbiolo, 2021-01-11:
> I think we can start moving small chunks of code to both swift/kotlin. How do you think we can start doing this?
Generally speaking, it is best to have a defined roadmap for all the changes that should made for v4.0 for example, so we don't rewrite code that won't be used anymore. When rewriting existing files, it is best to start from isolated services/utilities and work up into the modules exposed to React Native. 

Compatibility between Java/Kotlin is straightforward. Between Objective-C and Swift it gets a little bit complicated with things like `@objc` and linking but still it is fairly easy. The SDKs remain the same so it's only a language rewrite, with very low risk of regressions.

CodingWithNobody, 2023-04-15:
Check this https://www.youtube.com/watch?v=8QW5HDa4lQU&list=PLQhQEGkwKZUrempLnmxjt7ZCZJu1W3p2i&index=67