react-native-camera/react-native-camera #1034: stopPreview is not a function

jzhw0130, 2017-12-11:
> 
  _takePicture() {
    const options = {};
    this.camera.capture({metadata: options})
    .then((data) => {
      console.log(data.path)
      this.camera.stopPreview();          
    })
    .catch(err => console.error(err));
  }

> 
capture() works well, but stopPreview() failed, something wrong with my code?

sibelius, 2017-12-11:
@jzhw0130 this is only on master

we need a new release

can you try in master?

jzhw0130, 2017-12-11:
@sibelius 
Thanks

It works well for iPhone.
When will the feature release?

sibelius, 2017-12-11:
@rt2zz can we plan a new release?

jzhw0130, 2017-12-18:
@sibelius @rt2zz 

Any plan?

chris-247-sloan, 2017-12-19:
I'm getting the same issue. 

> can you try in master?

How do I try in master? 

package.json dependencies looks like this - 
```

"dependencies": {
    "native-base": "^2.3.5",
    "react": "16.0.0",
    "react-native": "0.51.0",
    "react-native-camera": "^0.12.0",
    "react-native-elements": "^0.18.5",
    "react-native-fingerprint-scanner": "^2.2.1",
    "react-native-vector-icons": "^4.4.2",
    "react-navigation": "^1.0.0-beta.21"
  },
```

Cheers.

jzhw0130, 2017-12-20:
Just copy master code to your project.

Nauzer, 2017-12-20:
Aah - that's why It's not working for me - good to know.
Any idea when this is going to be released?

chris-247-sloan, 2017-12-20:
Thanks.
I'm new to node, npm and react-native, Is there a definitive list somewhere of what can go into a package.json file?

jzhw0130, 2017-12-20:
@chris-247-sloan 

You want to know where to add the dependencies or how to edit the packet.json ?

chris-247-sloan, 2017-12-20:
@jzhw0130  I'm looking for documentation on package.json to understand better what possible entries it can have. It seems like such a big part of each project and I've yet to come across detailed documentation on it.

Any idea where i can find some?
Cheers

jzhw0130, 2017-12-21:
@chris-247-sloan 

I'm not sure. I'm new, too.
Maybe you can read: https://viblo.asia/p/understanding-the-contents-of-packagejson-YWOZrM3YKQ0

sibelius, 2018-01-07:
we gonna release soon, for now use master version

closed 2018-01-07 by sibelius

vrMars, 2018-01-15:
Any updates on the release?