react-native-camera/react-native-camera #1037: Build failed after upgrading to React Native v0.51.0

ikhsanalatsary, 2017-12-12:
This problem after upgrading my project dependencies to react-native v0.51.0, react v16.0.0, react-native-camera v0.12.0 & try building iOS in Xcode v8.2.1. Screenshots:
[image]

how to resolve this problem? 

Thank you

sibelius, 2017-12-12:
check the breaking changes section of rn51

ikhsanalatsary, 2017-12-12:
Consider negative border radii and widths invalid (99d294b) - @rsnara

[image]

ikhsanalatsary, 2017-12-15:
@sibelius apparently, the problem occurs from cocoapods. Before I upgrade RN version to v0.51. I did unlink react-native-camera & linking again but I don't know it linked to pods. So i remove it from pods and linking to my *.xcodeproj.

sibelius, 2017-12-15:
Great to hear that

closed 2017-12-15 by sibelius