react-native-camera/react-native-camera #1038: Losing focus Samsung A6

joelbarbosa, 2017-12-12:
On my Tablet Samsung A6, its losing auto focus,
So on this class: RCTCameraViewFinder.class
```java

// RCTCameraViewFinder.class line 411
public void handleFocus(MotionEvent event, Camera.Parameters params) {
....
 try {
                _camera.autoFocus(new Camera.AutoFocusCallback() {
                    @Override
                    public void onAutoFocus(boolean success, Camera camera) {
                        if (success) {
                             // this line is canceling my focus on Samsung only
                             // I coment it and run.
                            // camera.cancelAutoFocus(); 
                        }
                    }
                });
            } catch (Exception e) {
                e.printStackTrace();
            }
}
```
Any suggestions on this?

ilsonlasmar, 2018-04-05:
same problem. Any other solution?

joelbarbosa, 2018-04-05:
At moment I don't know. I used it temporary solution and the problem go away until we have a new fix update.

ilsonlasmar, 2018-04-05:
thanks @joelbarbosa! implemented your solution and it worked!

ruipaulo, 2018-06-07:
Any ideas how to apply this fix to the new RNCamera?

joelbarbosa, 2018-06-08:
@ruipaulo I think until we do not have a solution, you can try using the same approach on the 
method: public void handleFocus(MotionEvent event, Camera.Parameters params);

closed 2019-06-08 by stale[bot]