react-native-camera/react-native-camera #1040: [metadataOutput setMetadataObjectTypes:self.barCodeTypes]; fails in Debug Run iOS

tfsjohan, 2017-12-13:
https://github.com/lwansbrough/react-native-camera/blob/874a29f7cba278e875338cd1396889d75c12583c/ios/RCTCameraManager.m#L475

This line fails when running on iOS from Xcode in Debug Mode on real device. 

This worked previously, but I recently upgraded React Native and after that it broke. If I comment out this line I don't get errors, but I don't get any camera preview.

ngo275, 2018-03-22:
I deleted `barCodeTypes={['qr']}` in <Camera /> property, and it worked well.

export-mike, 2018-08-15:
@ngo275 are you using 0.10.0? I'm having the same issue. Once i've removed barCodeTypes={['qr']} it no longer scans for QR codes. I've downgraded as I'm having issues installing this library for android, I'm going in circles

closed 2019-06-08 by stale[bot]