react-native-camera/react-native-camera #1043: 5% crash rate on android (~2k daily active users)

seblau, 2017-12-14:
I am using `react-native-camera` to scan qr codes. The crashes appear across all android versions.

react-native: 0.43.4
react-native-camera: 0.10.0

Copied from google play console crash reports:
```
java.lang.RuntimeException: 
  at android.os.AsyncTask$3.done (AsyncTask.java:309)
  at java.util.concurrent.FutureTask.finishCompletion (FutureTask.java:354)
  at java.util.concurrent.FutureTask.setException (FutureTask.java:223)
  at java.util.concurrent.FutureTask.run (FutureTask.java:242)
  at android.os.AsyncTask$SerialExecutor$1.run (AsyncTask.java:234)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1113)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:588)
  at java.lang.Thread.run (Thread.java:818)

Caused by: java.lang.RuntimeException: 
  at android.hardware.Camera.native_getParameters (Native Method)
  at android.hardware.Camera.getParameters (Camera.java:1996)
  at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground (RCTCameraViewFinder.java:307)
  at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground (RCTCameraViewFinder.java:292)
  at android.os.AsyncTask$2.call (AsyncTask.java:295)
  at java.util.concurrent.FutureTask.run (FutureTask.java:237)
```

ssseasonnn, 2017-12-15:
same crash

sibelius, 2018-01-07:
check this new implementation https://github.com/react-native-community/react-native-camera/pull/1071

I hope it solves this

closed 2019-04-18 by sibelius