react-native-camera/react-native-camera #1044: Issue when no focus on Android

viacheslav-gotsuliak, 2017-12-14:
Hi there.

We're having strange issue with react-native-camera on Android. 
Capture promise never resolves when there is no focus on Android devices. To reproduce this issue it's enough to close camera and take photo. After that promise doesn't resolve (to then or error) even on photo with focus.

Maybe someone had something similar.

eugene-zheng0208, 2017-12-16:
+1
same issue for me

naddeoa, 2017-12-25:
I was able to debug this a little bit and narrowed it down to this line:

https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java#L559

Looks like there is a variable that needs to be switched but isn't for some reason.

naddeoa, 2017-12-25:
Pretty sure it was caused by this commit:

https://github.com/lwansbrough/react-native-camera/commit/12afd8443ac3bb0aaaca17f2a11c349d81e465d1

Looks like there is some sequence of events that can result in the onPicuteTaken callback not getting called. A few other posts online about it, like this one: https://stackoverflow.com/questions/24917850/onpicturetaken-never-called

We'll need a better way to throttle the capture requests that doesn't involve setting a variable in a callback. If that was the intent.

closed 2019-04-18 by sibelius