react-native-camera/react-native-camera #1048: Preview freezes after first Camera.capture on Android

pi-sw, 2017-12-15:
Hi there!
I experience an issue with the react-native-camera: the preview freezes after the first `camera.capture(...)` with the newest version 0.12.0 on Android 6.0.1, device: SM-G00F.

The following code is invoked, however - the `capture` method neither yields any data nor fails.
```
      const options = {
        audio: false,
        mode: Camera.constants.CaptureMode.still,
      };
      this.camera.capture({ metadata: options })
        .then((data) => {
          this.updatePhoto(data.path);
        })
        .catch((err) => {
          console.error(err);
        });
```

karltaylor, 2017-12-18:
Related: https://github.com/lwansbrough/react-native-camera/issues/1008

karltaylor, 2017-12-19:
@pi-sw, I had this problem when the `<Camera />` element was being unredered before it could complete. Are you sure you're not updating the state and unrendering the camera?

markoi, 2018-01-12:
Hi 

i have the same problem.. freezing of the preview window after camera.capture(); 

I noticed in adb logcat the following message after calling .capture()

// the logs are from a emulator... but i s the same on a real device...

_V EmulatedCamera_Camera: doStopPreview_

When I put the app in background and then again in foreground then there is a

_V EmulatedCamera_Preview: startPreview_

and the preview is working again... until the next .capture() call...

What could be wrong.. the camera element is rendered only once.

I tried to re-render the cam element and it works.. but the user can see the camera disappearing/blinking

sibelius, 2018-01-12:
try the experimental RNCamera module on android https://github.com/react-native-community/react-native-camera#experimental

markoi, 2018-01-12:
Oookey I take it back... 0.12.0 is working perfectly ...

I found that we switched from this react-native-camera module to another version because we needed  the barcode finder. 

"react-native-camera": "github:fbacker/react-native-camera#barcode-finder",

so I switched back to
"react-native-camera": "^0.12.0",

and is working as expected....ammm is there any barcode finder view in this module (that is not just a rendered rectangle but is also searching for barcodes inside the confined area... :P ... just asking )

tried with https://github.com/sahir/react-native-barcode-reader ... and it has the same problem as before with the preview freezing

thank you for the help... sorry I missed the correct module :P

wbroek, 2018-01-17:
@sibelius tried it on 0.12.0, 0.13.0 and experimental and only the 0.12.0 works. Meaning the preview windows stops streaming the live footage. If you do a capture again the resulting image is correct and showing the live footage.

sibelius, 2018-01-17:
could you help us figure it which commit this broke?

markoi, 2018-01-17:
I found in RCTCameraModule.java if I restored in the method captureWithOrientation(...)

this line 
camera.takePicture(**shutterCallback**, null, captureCallback);

to
camera.takePicture(**null**, null, captureCallback);

then everything works as expected

sibelius, 2018-01-17:
great, can you help a PR ?

then we can all test and merge in master again

wbroek, 2018-01-18:
@sibelius create a PR, change of the ShutterCallBack was introduced in https://github.com/react-native-community/react-native-camera/commit/874a29f7cba278e875338cd1396889d75c12583c#diff-c0b5ae039b51e601ab6eba76124ab4c2
Think it was for the barcode reading and not for taking normal photos.

Tested my PR on barcodes and normal photo taking

alexanderwallin, 2018-04-15:
I circumvented this by wrapping `<Camera />` inside a wrapper component that forces rendering of a new `<Camera />` to be rendered whenever an `updater` prop updates. Whenever that `updater` changes, the wrapper component does something like this:

```js
this.setState({ shouldRenderCamera: false }, () => {
  setTimeout(() => this.setState({ shouldRenderCamera: true }), 10)
})
```

And then, inside `render()`:

```js
render() {
  if (this.state.shouldRenderCamera === false) {
    return null
  }

  // Render <Camera /> here...
}
```

ithieund, 2018-04-20:
@alexanderwallin ,
Can you give an example code please?

Thanks.

alexanderwallin, 2018-04-20:
@ithieund 

```js
import React, { PureComponent } from 'react'
import PropTypes from 'prop-types'
import Camera from 'react-native-camera'

/**
 * Rerenderable Camera
 *
 * This wisardry forces a re-render of the <Camera /> plugin,
 * since v0.13.0 does not have a working startPreview() feature.
 */
class RerenderableCamera extends PureComponent {
  static propTypes = {
    cameraRef: PropTypes.func.isRequired,
    updater: PropTypes.number.isRequired,
  }

  state = {
    isVisible: true,
  }

  componentWillReceiveProps(nextProps) {
    const { updater } = this.props

    if (nextProps.updater !== updater) {
      this.setState({ isVisible: false })
    }
  }

  componentDidUpdate(prevProps) {
    const { updater } = this.props

    if (prevProps.updater !== updater) {
      setTimeout(() => this.setState({ isVisible: true }), 10)
    }
  }

  render() {
    const { cameraRef, ...props } = this.props
    const { isVisible } = this.state

    if (isVisible === false) {
      return null
    }

    return <Camera ref={cameraRef} {...props} />
  }
}

export default RerenderableCamera

```

And then I use it like this:

```js
<RerenderableCamera
  cameraRef={camera => {
    this.camera = camera
  }}
  updater={lastRetakeAt}
  aspect={Camera.constants.Aspect.fill}
  orientation={Camera.constants.Orientation.landscapeLeft}
  captureTarget={Camera.constants.CaptureTarget.disk}
  type={Camera.constants.Type.front}
/>
```

where `lastRetakeAt` is a numeric timestamp.

ios-dev-newbie, 2018-07-03:
I have used the above Rerenderable camera but getting the issue as cameraRef undefined.

alexanderwallin, 2018-08-02:
@ios-dev-newbie Are you passing the `cameraRef` prop a function as in the example above?

sibelius, 2018-08-02:
@alexanderwallin we have start and stop preview methods on latest version and master

can you check?

is this still a problem?

alexanderwallin, 2018-08-02:
Sorry, won't be able to do any checking on this!

SimonErm, 2018-09-25:
RCTCamera is deprecated, use RNCamera instead. Please open a new issue if this problem still occures.

closed 2018-09-25 by SimonErm