react-native-camera/react-native-camera #1050: Multiple Camera Preview Stream

rubencg195, 2017-12-16:
I am currently wanting to implement a View where I can show two instances of the <Camera> component. One as the original camera preview and the second one as a secondary component to show filters and object recognition implementation for which I need to have access to the camera stream in real time for image-processing. 

Using a unique component there is no problem, but when trying to add the second one, both view shows one of them as a black screen. Also, I haven't find a way to have access to the camera stream for image-processing. 

Note: Both using a dummy camera on an emulator. and the same issue on a real device.

1) Using an unique camera component:

[image]

2) Using two camera components:

[image]

mateusz-, 2018-02-09:
This can also happen when having more than one camera component across scenes when using a router (in my example, react-native-router-flux).

sibelius, 2018-02-09:
fixed on master with RNCamera

closed 2018-02-09 by sibelius

rubencg195, 2018-02-09:
How can I do to show my camera in two diferentes components, I want to do an app compatible with Google cardboard for augmented reality.

mateusz-, 2018-02-09:
Thanks @sibelius.

For anyone slightly confused by the update (like me), update your Camera component to [RNCamera](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md) and all should be good.

MrPluto, 2018-06-13:
Hi @rubencg195 
how do you do **real time image-processing** or **image filter** by using this library?
 can you give me some tips