react-native-camera/react-native-camera #1051: Camera in not unmounted when navigating to a different screen.

priyankverma, 2017-12-18:
Hello, i am using:

"react-native": "0.48.3",
"react-native-camera": "^0.12.0",
"react-native-router-flux": "^3.41.0"

And the problem is whenever i navigate from one screen to other, the camera is not unmounted in the background, please help it out.

jpudysz, 2017-12-29:
Use https://www.npmjs.com/package/@patwoz/react-navigation-is-focused-hoc and unmount it via JSX

`{this.props.isFocused ? this.renderCamera() : null}`

xiamx, 2017-12-30:
It's probably better to have a solution native to react-native-camera

ospfranco, 2018-01-13:
this is a navigation issue and heavily project dependent, make sure you unmount your containers.

closed 2018-01-13 by ospfranco

Bartlebyy, 2018-10-08:
I had the same problem. The camera component would not unmount when I switched between tabs. So I took advantage of the `componentDidMount` and added two listeners `willFocus` and `willBlur` and now it works. 
``` 
 componentDidMount() {
    const { navigation } = this.props;
    navigation.addListener('willFocus', () =>
      this.setState({ focusedScreen: true })
    );
    navigation.addListener('willBlur', () =>
      this.setState({ focusedScreen: false })
    );
  }

  render() {
    const { hasCameraPermission, focusedScreen } = this.state;
    if (hasCameraPermission === null) {
      return <View />;
    } else if (hasCameraPermission === false) {
      return <Text>No access to camera</Text>;
    } else if (focusedScreen){
      return (this.cameraView());
    } else {
      return <View />;
    }
  }
```

hladf, 2019-10-08:
I'm getting the same error... i'd tryied all this solutions and **nothing** works. Any Idea to 're-mount' the camera component?

RishavKumar-3796, 2019-12-04:
Why the hell is this issue closed?

tapsygroup, 2020-11-26:
this works for me. 
if you are using a different screen for CAMERA, you can easy unmount it when moving to another screen.

<Stack.Screen name="camera" component={CameraScreen} options={{unmountOnBlur: true}}/>
        </Stack.Navigator>

DanielMarinado, 2022-05-27:
> this works for me. ( Navigation 5.x ) if you are using a different screen for CAMERA, you can easy unmount it when moving to another screen.
> 
> <Stack.Screen name="camera" component={CameraScreen} options={{unmountOnBlur: true}}/> </Stack.Navigator>

Thanks a lot! Works for me!

lehigh123, 2023-04-11:
related stack overflow question https://stackoverflow.com/questions/75987336/app-crash-when-expo-camera-view-unmounted