react-native-camera/react-native-camera #1053: Orientation of captured photo wrong

vvusts, 2017-12-20:
I implemented camera like in documentation.
If I take photo with camera and phone is landscape (lock button up) image orientation is good.
But if I do the same with lock button down image is turned upside down.
On android work fine this issue is on IOS.

Does anybody have any idea how to fix this?

almasaeed2010, 2018-01-19:
Hello,

We are also experiencing the same problem where the orientation on the browser is not the same as the phone.

Problem is occurring in iOS 11.
Locking the orientation to "portrait" seems to work but setting it to "auto" unfortunately breaks it.

Thanks for the help!

tirrorex, 2018-01-22:
See https://github.com/react-native-community/react-native-camera/issues/1102 for answer

almasaeed2010, 2018-01-22:
Thank you for pointing that out!

tirrorex, 2018-01-22:
No problem mate, we actually had the same issue with another app (images uploaded to our backend server).
You can refer to this thread on SO for more informations : https://stackoverflow.com/questions/10850184/ios-image-get-rotated-90-degree-after-saved-as-png-representation-data

I think this issue can occur on android too depending on what device you have.

zedtux, 2018-07-03:
I can confirm @tirrorex that it occurs on my Android phone (Xiaomi Redmi).

n1ru4l, 2018-12-13:
:tada: This issue has been resolved in version 1.6.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v1.6.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

mcorner, 2018-12-14:
I don't think this is fixed.  See my comment in #1807.

I am using 1.6.3 and:
      const options = { quality: 0.5, base64: true, orientation: RNCamera.Constants.ORIENTATION_UP };
      const data = await this.camera.takePictureAsync(options);

AUTO doesn't work either.  It is just on samsung phones.

n1ru4l, 2018-12-14:
@mcorner does it also occur when the image is not returned as base64?

n1ru4l, 2018-12-14:
Also it should be orientation: 'portrait' check the docs pls

mcorner, 2018-12-14:
"portrait" produces the same result.  Removing base64 is a bit harder because of our code, but I can try that if you think that will matter.

mcorner, 2018-12-14:
We are uploading the file from storage to s3, and that image is rotated as well, so it isn't just the base64.

sibelius, 2018-12-14:
You can save to disk and upload directly using Blob, it would be faster than using base64

mcorner, 2018-12-14:
We are.  We are just using the base64 as a convenient preview.

n1ru4l, 2018-12-14:
@mcorner I have no Samsung test device. It is working as expected on my nexus 5x device. Could you please dig into whats the issue and fix it? Possibly related: https://github.com/google/cameraview/issues/22

sibelius, 2018-12-14:
I think we need a testing matrix for orientation

We have front and back camera
We have many device orientations
We have many display orientations
We have many options to try to fix them

We need to understand in which combination is failing

sibelius, 2018-12-14:
Much like this comment https://github.com/react-native-community/react-native-camera/pull/1992#issuecomment-447011051

mcorner, 2018-12-14:
I will get a Samsung device to test with (it doesn't happen on my pixel).  The fixOrientation does work, it is just abysmally slow.  My best option ATM is to hide that latency with a UI element (spinner).  I am bit surprised that rotating a bitmap is this slow, Expo seems to be using the same exact method to do it (createBitmap: [expo](https://github.com/expo/expo/blob/1add34f1687336787591dcc4144a7f5e4f361e79/android/versioned-abis/expoview-abi31_0_0/src/main/java/abi31_0_0/expo/modules/camera/tasks/ResolveTakenPictureAsyncTask.java#L251))

mcorner, 2018-12-14:
Note that this is actually a problem on iOS as well...  It takes a photo containing exif rotation data.  So if you put this into a uiwebview etc that does not know how to deal with exit rotation it shows up as rotated incorrectly.  The docs say that "fixOrientation" is Android only, but the ios Code seems to respect the option: 

https://github.com/react-native-community/react-native-camera/blob/a5a93a73c29bdeced8620bb71c0b86ce49a86f1d/ios/RCT/RCTCameraManager.m#L667

sibelius, 2018-12-14:
for fast android image take picture, we need this:

https://github.com/react-native-community/react-native-camera/issues/1836

n1ru4l, 2018-12-14:
I am also not experiencing this issue on iOS I am confused.

mcorner, 2018-12-16:
After some testing, this is my understanding:

The photos taken with RNCamera iOS (and Samsung phones) are rotated 90degrees CCW.  To compensate for that the exif includes an Orientation tag of "6", meaning "please rotate this image 90 degrees Clockwise when you display it".  Many things deal with it correctly, like chrome and gmail.  But many places seem not to, like a UIWebView in an android app, Slack, or even jsfiddle in Chrome.

Up arrow: https://s3.amazonaws.com/petcarenow-store/staging/chatMedia/LJt4-241935b1-2622-4ead-bfd0-f58c477510eb.jpg was taken with RNCamera.  In Chrome and Safari it is up as it should be.  In a UIWebview on Android it is rotated 90CCW.  

Or in jsfiddle it is wrong also:
https://jsfiddle.net/4gpx02q9/4/

If you paste that image into slack it is correct on the desktop and rotated on an ios or an android phone. (super odd).

So the iOS and Samsung images are correct in places that respect exif when displaying, but many places do not.  The fixOrientation rotates it and erases the exif Orientation data which makes it display correctly always.  It should probably be the default.

But...for some reason fixOrientation doesn't work for me on iOS even though the code is there.  I will dig into that next week to see if I can get that to work.

mcorner, 2018-12-16:
forceUpOrientation: true seems to work on iOS for me to get the rotation right.

Example: https://s3.amazonaws.com/petcarenow-store/development/chatMedia/LJt4-9ed82f44-51e7-4518-8a0c-2f03378a1740.jpg

The exif is: 
Orientation                     : Horizontal (normal)

arneson, 2018-12-18:
@mcorner 
Experiencing the same on Nokia devices.
You got anywhere on Android?

mcorner, 2018-12-18:
> @mcorner
> Experiencing the same on Nokia devices.
> You get anywhere on Android?

On Android use: fixOrientation . That will rotate the image and erase the exif rotation.  It may be a bit slow though.

arneson, 2018-12-18:
@mcorner 
Thanks, I had the wrong name of the flag, found forceUpOrientation somewhere but now it works. 
Yeah, it sure is slow. I guess I have to add a spinner and freeze the rest until this is fixed. Slightly annoying.

vinayan3, 2018-12-18:
I'm having similar problems where `fixOrientation` is incredibly slow on 
* Nexus 5X - 8.1.0. ~ 7/8 seconds
* Samsung SM-G925A  - 7.0 ~4/5seconds

The code I'm using are:
```
        const options = {
          quality: 1,
          orientation: "portrait",
          pauseAfterCapture: true,
          fixOrientation: true
        }
        const image = await this.camera.takePictureAsync(options)
```

n1ru4l, 2018-12-19:
@vinayan3 Sadly there is currently no more performant implementation of rotating a Bitmap that I am aware of. If you have contact to any experienced android developers or are one yourself it would be awesome if someone could help out with this issue.

arneson, 2018-12-19:
What about doing what iOS and some cameras do? Don’t rotate it, but set the exif orientation value?

n1ru4l, 2018-12-19:
@arneson Please read https://github.com/react-native-community/react-native-camera/issues/1053#issuecomment-447615936

mcorner, 2018-12-19:
https://github.com/CameraKit/camerakit-android seems to use JNI to libjpeg-turbo for rotation.  A bit of a project unfortunately.

sibelius, 2018-12-19:
@mcorner feel free to send a pr to fix this

vinayan3, 2019-01-05:
@n1ru4l I've going through the code and on Android if there is any post processing what happens is 
there is no Exif data being included into the output JPEG.  I hacked up some code to include the ExifData in `ResolveTakenPictureAsyncTask.java`.
```
    @Override
    protected WritableMap doInBackground(Void... voids) {
            // Bunch of code later...
            // [ . . . ] 
            // Write compressed image to file in cache directory unless otherwise specified
            if (!mOptions.hasKey("doNotSave") || !mOptions.getBoolean("doNotSave")) {
                String filePath = writeStreamToFile(imageStream);
                if (exifInterface != null) {
                    ExifInterface outExif = new ExifInterface(filePath);
                    RNCameraViewHelper.setExifData(exifInterface, outExif);
                    RNCameraViewHelper.setExifHeightWidth(
                        outExif,
                        mBitmap.getHeight(),
                        mBitmap.getWidth());
                    outExif.saveAttributes();
                }
                File imageFile = new File(filePath);
                String fileUri = Uri.fromFile(imageFile).toString();
                response.putString("uri", fileUri);
            }
```
If you read the docs about `ExifInterfance` it indicates that `saveAttributes()` is slow because it reads the file, writes the exif data in a temp file, and then deletes the original final. :/.  If anyone thinks this would be useful or the correct solution I can get a PR Together for this.

zedtux, 2019-06-01:
Yeah, well done bot 👏. Useful work.

fabriziobertoglio1987, 2019-07-17:
I am experiencing this issue on an ios device running iOS 10.3.3
I also updated to the latest react native camera version 2.11.1, which added the extif feature for checking camera orientation on android, but I believe this does not relate to iphone. The device is an old iphone 5. I could not extensively test this feature on an iphone 10, but on the emulator seems to work on iphone 10.

fabriziobertoglio1987, 2019-07-17:
> @n1ru4l I've going through the code and on Android if there is post processing, `skipProcessing` is not set, what happens is
> there is no Exif data being included into the output JPEG. The call to reproduce this behavior is
> 
> ```
>         const options = {
>           quality: 1,
>           orientation: "portrait",
>           pauseAfterCapture: true,
>           fixOrientation: true
>         }
>        camera.takePicture(options)
> ```
> 
> I hacked up some code to include the ExifData in `ResolveTakenPictureAsyncTask.java`.
> 
> ```
>     @Override
>     protected WritableMap doInBackground(Void... voids) {
>             // Bunch of code later...
>             // [ . . . ] 
>             // Write compressed image to file in cache directory unless otherwise specified
>             if (!mOptions.hasKey("doNotSave") || !mOptions.getBoolean("doNotSave")) {
>                 String filePath = writeStreamToFile(imageStream);
>                 if (exifInterface != null) {
>                     ExifInterface outExif = new ExifInterface(filePath);
>                     RNCameraViewHelper.setExifData(exifInterface, outExif);
>                     RNCameraViewHelper.setExifHeightWidth(
>                         outExif,
>                         mBitmap.getHeight(),
>                         mBitmap.getWidth());
>                     outExif.saveAttributes();
>                 }
>                 File imageFile = new File(filePath);
>                 String fileUri = Uri.fromFile(imageFile).toString();
>                 response.putString("uri", fileUri);
>             }
> ```
> 
> If you read the docs about `ExifInterfance` it indicates that `saveAttributes()` is slow because it reads the file, writes the exif data in a temp file, and then deletes the original final. :/. If anyone thinks this would be useful or the correct solution I can get a PR Together for this.

has this been solved with the following update of [react-native-camera version 2.9](https://github.com/react-native-community/react-native-camera/releases/tag/v2.9.0) ?

vinayan3, 2019-07-17:
@fabriziobertoglio1987 I haven't upgraded so I may not be able to tell you. Sorry for not being helpful.

CareerFairPlus, 2019-07-23:
Using version 2.11.1, using the following options:
```    
  const options = {
        quality: 0.7,
        forceUpOrientation: true,
        orientation: 'portrait',
        pauseAfterCapture: true,
      };
```

I get the correct output on iOS and on the Nexus 5X.  However when testing on the Pixel 2 XL, the image is rotated and doesn't seem to have the EXIF data included in the file because when viewing the file on a browser, it's still rotated.  If I include `fixOrientation: true` in the options, then I get the correct output on the Nexus 5X and Pixel 2 XL, however the Nexus 5X is extremely slow.  I don't want to use that option if I don't have to on the Nexus 5X.  But I'm stuck because the Pixel 2 XL doesn't include the proper EXIF data.

CareerFairPlus, 2019-07-23:
It seems to be that just the Orientation attribute in the EXIF data is not being set correctly.  The issue is with setExifData() function in RNCameraViewHelper.java.  It's assuming all Numbers are Doubles, but setting a double to the Orientation attribute results in an unknown Orientation.  I coded up a quick fix for this for my use case, but I'm not sure what other attributes are Integers:
```
  public static void setExifData(ExifInterface exifInterface, WritableMap exifMap) {
    ReadableMapKeySetIterator iterator = exifMap.keySetIterator();
    while (iterator.hasNextKey()) {
      String key = iterator.nextKey();
      switch (exifMap.getType(key)) {
        case Null:
          exifInterface.setAttribute(key, null);
          break;
        case Boolean:
          exifInterface.setAttribute(key, Boolean.toString(exifMap.getBoolean(key)));
          break;
        case Number:
          // This is to fix an issue with using doubles for Orientation
          if (key.equals(ExifInterface.TAG_ORIENTATION)) {
            exifInterface.setAttribute(key, Integer.toString(exifMap.getInt(key)));
          } else {
            exifInterface.setAttribute(key, Double.toString(exifMap.getDouble(key)));
          }
          break;
        case String:
          exifInterface.setAttribute(key, exifMap.getString(key));
          break;
      }
    }
  }
```

AMINEDGE, 2019-07-23:
there is also a `mirrorImage` parameter you can pass as `options` which takes a `boolean` and flips the image if given value is true, but it always flips the image which is not needed, you can use it with some conditions

CareerFairPlus, 2019-07-24:
@AMINEDGE Unfortunately that won't help for this issue seeing as the topic is about rotation (Orientation), not about mirroring on the vertical axis.

zedtux, 2019-09-23:
Anyone can confirm the fix from #2396 works fine?

zedtux, 2019-11-22:
Chuuuut

fabriciosautner, 2019-12-19:
Has anyone found a solution that works for iOS (and doesn't spoil the Android setup)?

zedtux, 2020-02-18:
bla

tyleralves, 2020-04-24:
We were seeing this issue on one of our test devices (Galaxy S9) on RNCamera@3.8.0 (Oct 2019), but we are unable to reproduce since updating to 3.21.0. It would be amazing to be able to remove the `fixOrientation` flag (for obvious reasons), but we are reluctant since I don't see any code changes to the library that would have fixed this between these versions.

I would love to hear if people are still experiencing this, and which devices have to issue.

tyleralves, 2020-04-30:
Pinging people active in this discussion because we have tested quite a bit and it looks fixed. This is only based on our one Samsung S9, and there doesn't seem to be any explicit fix committed here.

@fabriziobertoglio1987 @zedtux @arneson @mcorner

zedtux, 2020-05-01:
Thank you @tyleralves for pinging me, I don't have time to check that now and for a while, sorry.

closed 2020-07-08 by stale[bot]

dancherb, 2021-03-31:
Still getting this issue on Android emulator (using webcam input), even fixOrientation does not work.

ilyasmussin, 2022-11-16:
Still experience the same issue when users "take pictures" with their iPhones and view it later in iOS Safari. The issue persists when testing with or without the heic2any library. Would appreciate any suggestions.

PR #1992 (2018-12-13): fix(android): use device orientation for images/videos instead of display orientation
This is my attempt for fixing #1725 based on the work of @Fsarmento (https://github.com/react-native-community/react-native-camera/issues/1725#issuecomment-441599916 https://github.com/react-native-community/react-native-camera/compare/master...Fsarmento:master)

This PR unifies the orientation used for the pictures/video with the `ios` implementation by using the device orientation instead of the display orientation.

**Related:**

Fixes #1725
Closes #1941
Fixes #1053

cc @sibelius @JulienUsson @SimonErm @Ts [... cut]