react-native-camera/react-native-camera #1054: Android - camera using onBarCodeRead crashes app on second instance

eltanno, 2017-12-20:
We are using the camera in one screen of a multi screen app on Android. If you add a onBarCodeRead handler to the camera instance the first time you load the camrea screen it is fine. The second time you load the screen the app crashes. This is a bug in the Java code. In RTCCameraViewFinder this code results in a crash as the camera.getParameters() is unpopulated. I think the previous instance of the camera isnt released properly. If you move these lines inside the try / catch block everything works fine.  

RTCCameraViewFinder.java line 309
```
                Camera.Size size = camera.getParameters().getPreviewSize();
                int x = 0;
                int y = 0;
                int width = size.width;
                int height = size.height;
```
I will submit a PR with the code change.

eltanno, 2017-12-20:
Looks like the master branch has this fix in already, but not the NPM released version

charleycesar, 2018-02-20:
when it will be publish??? It's happen with me too

sibelius, 2018-02-20:
Fixed on latest release

closed 2018-02-20 by sibelius