react-native-camera/react-native-camera #1067: Сapture promise never resolves on Android.

VTSingle, 2017-12-31:
Body: 
using the example from github 

takePicture() {
    const options = {};
    this.camera.capture({metadata: options})
      .then((data) => console.log(data))
      .catch(err => console.error(err));
  }

Invoking takePicture function on some Android devices triggers neither 'then' nor 'catch' callbeck. 
The problem is faced on xiaomi and Meizu phones. Permissions for camera use were added to AndroisManifest. 
Interestingly, but in debug mode this problem doesn't happen. 

On Samsung S8, after some times of using camera capture the application is crashed.

kumarvish, 2018-01-02:
Are you making any changes to local state in this component? I experienced this behavior if there was re-rendering due to change in local state.

closed 2018-01-04 by VTSingle

PanRada, 2018-01-27:
Have issues on similar problem with promise.

ranjitnadar, 2018-01-30:
any solution for this promises issue on android

dann1609, 2018-03-19:
I has this problem to, but only on android. On IOS it works very well!

I tried changing etccamera to RNCamera but issue still present (Problem is bigger in RN)