react-native-camera/react-native-camera #1074: No callback when using front camera on Samsung S8

tommynordli, 2018-01-04:
I'm having issues with taking photos using `type={Camera.constants.Type.front}`. No callback is called on `capture()`, so I'm not getting any data or error. I haven't been able to find out why it happens, but out of all the phones I've tested it on it only happens on Samsung S8 using the front camera. It's not happening all the time, but most of the times. Back camera is working fine.

tommynordli, 2018-01-04:
I was using the module with commits after latest release. The issue isn't present on v0.12.0, so one of the latest commits on master has to be the problem. Closing this issue.

closed 2018-01-04 by tommynordli

VTSingle, 2018-01-04:
I use S8
/////////////////

import Camera from 'react-native-camera';

this.state = {
            cameraType: 'front'
        }

 render() {
          return (
                        <Camera
                        ref={(cam) => {
                            this.camera = cam;
                        }}
                        style={styles.preview}
                        //  aspect={Camera.constants.Aspect.fill}
                        type={this.state.cameraType}
                        captureAudio={false}
                        playSoundOnCapture ={false}
                        captureQuality={Camera.constants.CaptureQuality["480p"]}
                        fixOrientation={true}
                        mirrorImage={true}
                        //captureTarget={Camera.constants.CaptureTarget.disk}
                    >
                    </Camera>
         )
}

hardcodet, 2018-01-13:
@VTSingle  Could you resolve that issue? Given Samsung's market share here, that would be a show stopper for us 😨

sibelius, 2018-01-13:
@tommynordli  in which commit or PR this broke?

sibelius, 2018-01-15:
would u mind sending a PR to revert this change on master?

so we can review again