react-native-camera/react-native-camera #1079: Camera not identifying any QR or Barcodes

lucasharada, 2018-01-08:
I followed all Usage steps to run this component in my application. Right now, I can see the Camera component in the app, but I can't make it read any QR or Barcodes. My code is below:

```
import React, { Component } from 'react';
import { Dimensions, View, Text, Image, TouchableOpacity } from 'react-native';
import { HeaderBackButton } from 'react-navigation';
import Camera from 'react-native-camera';

const { height, width } = Dimensions.get('window');

class PointCamera extends Component {

  static navigationOptions = ({ navigation }) => ({
    title: 'Get points',
  });

  constructor() {
    super();

    this.onBarCodeRead = this.onBarCodeRead.bind(this);
  }

  onBarCodeRead(e) {
    console.log(
        'Barcode Found!',
        `Type: ${e.type} \nData: ${e.data}`
    );
  }

  goToWriteCode() {
    const navigation = this.props.navigation;
    navigation.navigate('WriteCode');
  }

  renderQRCodeScanner() {
    let squareWidth = height * 0.75;

    if (height > width) {
      squareWidth = width * 0.75;
    }

    return (
      <View
        style={{ width: squareWidth, height: squareWidth, backgroundColor: 'green' }}
      >
        <Image
          style={{ flex: 1 }}
        />
        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={{ flex: 1, margin: 10 }}
          onBarCodeRead={this.onBarCodeRead.bind(this)}
          barCodeTypes={['qr']}
          aspect='fill'
        />

      </View>
    );
  }

	render() {
		return (
			<View style={{ flex: 1, backgroundColor: '#bbb' }}>
				<Text
					style={{
						fontSize: 24,
						color: 'white',
						alignSelf: 'center',
						fontWeight: 'bold',
            marginTop: 20
					}}
				>
					Read the code
				</Text>
        <View style={{ marginTop: 20, backgroundColor: 'darkgreen', alignItems: 'center' }}>
          {this.renderQRCodeScanner()}
        </View>
        <TouchableOpacity
          onPress={this.goToWriteCode.bind(this)}
          style={{
            flex: 1,
            justifyContent: 'flex-end',
            alignItems: 'center',
            paddingBottom: 10
          }}
        >
          <Text
            style={{
              fontSize: 16,
              color: 'white',
              alignSelf: 'center'
            }}
          >
            Camera doesn't work
            <Text style={{ textDecorationLine: 'underline' }}>
              {'\n'}insert manually.
            </Text>
          </Text>
        </TouchableOpacity>
			</View>
		);
	}
}

export default PointCamera;

```

sibelius, 2018-01-08:
can you provide a repo to easy reproduce this?

can you test with master version?

lucasharada, 2018-01-08:
@sibelius Unfortunatelly, I can't provide a repository for this. I tested with the master version and didn't get to work. I, then, proceeded to try with [QRCode Scanner](https://github.com/moaazsidat/react-native-qrcode-scanner) and worked.

Thanks for your time!

closed 2018-01-08 by lucasharada