react-native-camera/react-native-camera #1083: Access Error 'NSMicrophoneUsageDescription' without accessing to camera

kosiakMD, 2018-01-09:
"react": "16.0.0-alpha.12",
"react-native": "0.47.1",
"react-native-camera": "^0.12.0",

Everything was good but started receive error
```[access] This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSMicrophoneUsageDescription key with a string value explaining to the user how the app uses this data.```

Camera used for Barcode Reading only and didn't need access to microphone and other extra abilites

Did some policies changes in iOS? Or what went wrong?

mckn, 2018-01-09:
I guess you need to add the NSMicrophoneUsageDescription in info.plist

sibelius, 2018-01-09:
add this to your Info.plist

closed 2018-01-09 by sibelius

tonygentilcore, 2018-02-05:
@kosiakMD issue #446 has a better solution. To avoid need for mic permission, make sure to only call `checkVideoAuthorizationStatus` and not `checkDeviceAuthorizationStatus`.

janziemba, 2018-04-13:
You need to add all permission keys to Info.plist, even if you don't use them. See: https://github.com/yonahforst/react-native-permissions#app-store-submission-disclaimer

winkelsdorf, 2019-05-05:
@janziemba That's just for the library `react-native-permissions` which uses (checks) all of the listed permissions. Usually you don't have to add _all_ permission keys.