react-native-camera/react-native-camera #1086: Could not find com.github.react-native-community:cameraview:df60b07573

rcorrie, 2018-01-10:
I am getting the following error running `react-native run-android` with the latest code on master.

```
* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.github.react-native-community:cameraview:df60b07573.
     Required by:
         project :app > project :react-native-camera

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
```

sibelius, 2018-01-10:
Check README, now you need jitpack

closed 2018-01-10 by sibelius

nicemanis, 2018-01-15:
I added `maven { url "https://jitpack.io" }` to the android buildscript repositories

sibelius, 2018-01-15:
what is the error with jitpack?

nicemanis, 2018-01-15:
there is no error with jitpack, it works with react-native v0.52.0, but not with v0.51.0

sibelius, 2018-01-15:
What is the erro with 51?

nicemanis, 2018-01-15:
The same which opened this issue:
`Could not find com.github.react-native-community:cameraview:df60b07573`

It seems that rn v0.51.0 ignores the jitpack

nicemanis, 2018-01-15:
Ahh, the react-native-camera v0.6.0 also doesn't work on rn v0.52.0, I have the PropTypes error after successful build. At first I couldn't see it because rn v0.52.0 also breaks react-native-vector-icons

nicemanis, 2018-01-15:
Note that I'm forcing all subprojects to use android build tools version 25.0.2:
```
subprojects {
    afterEvaluate {project ->
        if (project.hasProperty("android")) {
            android {
                compileSdkVersion 25
                buildToolsVersion '25.0.2'
            }
        }
    }
}
```

sibelius, 2018-01-15:
We use android 26 now

nicemanis, 2018-01-15:
I checked the ./node_modules/react-native-camera/android/build.gradle:
```
buildscript {
  repositories {
    jcenter()
  }

  dependencies {
    classpath 'com.android.tools.build:gradle:1.2.3'
  }
}

apply plugin: 'com.android.library'

android {
  compileSdkVersion 23
  buildToolsVersion "23.0.1"

  defaultConfig {
    minSdkVersion 16
    targetSdkVersion 22
    versionCode 1
    versionName "1.0"
  }
  lintOptions {
    abortOnError false
    warning 'InvalidPackage'
  }
}

repositories {
  mavenCentral()
}

dependencies {
  compile "com.facebook.react:react-native:0.19.+"
  compile "com.google.zxing:core:3.2.1"
  compile "com.drewnoakes:metadata-extractor:2.9.1"
}
```

This is react-native-camera v0.6.0 and it shows 23.0.1 build tools version, is it supposed to be 26 ?

sibelius, 2018-01-15:
check RNCameraExample https://github.com/react-native-community/react-native-camera/blob/master/RNCameraExample/android/build.gradle#L27

you should use master branch: https://github.com/react-native-community/react-native-camera#how-to-use-master-branch

nicemanis, 2018-01-15:
Finally managed to get a working build... the problem was, that react-native-android android and google dependencies were conflicting with react-native-firebase, had to add this to `./android/app/gradle.build`:
```
compile (project(':react-native-camera')) {
    exclude group: "com.google.android.gms"
    exclude group: "com.android.support"
}
...
compile ('com.google.android.gms:play-services-vision:11.6.0') {
    force = true;
}

compile ('com.android.support:exifinterface:26.0.1') {
    force = true;
}
```
The error messages also didn't help - at first there was a build error that asked to add this to the AndroidManifest.xml:
```
<meta-data
    android:name="com.google.android.gms.version"
    android:value="@integer/google_play_services_version"/>
```
but if you added that, then it showed this error `Failed to initialize app. FirebaseOptions missing or invalid 'apikey' property.` which didn't make any sense. When I tryed to redo everything again, I noticed that there was a message that said that `com.android.support:exifinterface:26.0.1` was conflicting.

sibelius, 2018-01-15:
join expo slack on react-native-camera channel https://github.com/react-native-community/react-native-camera#contributing

and we will try to help you

nicemanis, 2018-01-15:
It's ok, I managed to fix this issue, everything is working now

sibelius, 2018-01-15:
Could u please update README to include all this information?

nicemanis, 2018-01-15:
Ok, I will update it as soon as I'll have some free time. I see that this issue is referenced [here](https://github.com/react-native-community/react-native-camera#meta-data-android-26), but as I posted before, this solution didn't work for me, so I can add the case description and solution for my situation. Also, the solution that is currently mentioned in the README doesn't work (at least it didn't for me) as you need to also add the actual meta-data tag which I mentioned previously. I can also add the meta-data tag example if necessary.

codecosine, 2018-01-17:
add this to ./android/app/gradle.build
its work for me without using force
```
compile (project(':react-native-camera')) {
        exclude group: "com.android.support"
        exclude group: "com.google.android.gms"
   }
```

daFelix, 2018-01-17:
Hello guys, Im kinda new into RN and got the same issue, following @nicemanis instruction excluding "com.android.support" and "com.google.android.gms" (since I imported react-native-firebase)  but it keep showing me that I need to add 

```
"< meta-data
    android:name=" com.google.android.gms.version "
    android:value=" @integer/google_play_services_version "/>"
```

in the AndroidManifest.xml 

Btw, Im using react-native-camera@0.6 since my RN is 0.50.4

"react": "16.0.0",
    "react-native": "0.50.4",
    "react-native-camera": "0.6",
    "react-native-config": "^0.10.0",
    "react-native-firebase": "^3.2.2"

Thanks in advance !

muratozgul, 2018-01-17:
Why is this closed? So what do I need to do to fix this error and build for Android. Which version of react-native, react-native-camera I need to fall back to. There is nothing clear in README about this error.

nicemanis, 2018-01-17:
@daFelix try using the master branch from git, edit the `package.json` to this `"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera"`

I'm using rn v0.51.0 and it's working fine

daFelix, 2018-01-17:
@nicemanis I  already edited the "react-native-camera:" to "git+https://git@github.com/react-native-community/react-native-camera"  and  upgraded to rn v0.51.0 , still nothing good :( 

here are my dependencies:

```
dependencies {

  compile (project(':react-native-camera')) {
     exclude group: "com.android.support"
    exclude group: "com.google.android.gms"
  }

  compile ('com.android.support:exifinterface:26.0.1') {
         force = true;
  }
  compile ('com.google.android.gms:play-services-vision:11.6.0') {
     force = true;
  }

    compile(project(':react-native-firebase')) {
        transitive = false
    }
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile 'com.android.support:appcompat-v7:26.0.1'
    compile "com.facebook.react:react-native:+"  // From node_modules
    compile ('com.google.android.gms:play-services-base:11.6.0') 
    compile "com.google.firebase:firebase-core:11.6.0"
    compile "com.google.firebase:firebase-database:11.6.0"
    compile project(':mapbox-react-native-mapbox-gl')

}
```

And here is part of my package.json dependencies: 

```
    "react": "16.0.0",
    "react-native": "0.51.0",
    "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera",
    "react-native-config": "^0.10.0",
    "react-native-firebase": "^3.2.2",
    "react-native-permissions": "^1.0.6",
    "react-navigation": "^1.0.0-beta.21",
```

It shows me this error: 

[image]

nicemanis, 2018-01-17:
@daFelix can you show your AndroidManifest.xml ?

daFelix, 2018-01-17:
@nicemanis  here it goes: 

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.interfasecity"
    android:versionCode="1"
    android:versionName="1.0"
     xmlns:tools="http://schemas.android.com/tools">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-feature android:name="android.hardware.camera" />
    <uses-feature android:name="android.hardware.camera.autofocus" />

    <uses-sdk
        android:minSdkVersion="16"
        android:targetSdkVersion="26" />

    <application
      android:name=".MainApplication"
      android:allowBackup="true"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:theme="@style/AppTheme"
      tools:node="replace">

      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
        android:windowSoftInputMode="adjustResize">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />
    </application>

</manifest>

```

nicemanis, 2018-01-17:
@daFelix  Try removing `xmlns:tools="http://schemas.android.com/tools"` and `tools:node="replace"` and rebuild

daFelix, 2018-01-17:
@nicemanis It show the previous error 

```
Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.1.0) from [com.android.support:appcompat-v7:26.1.
0] AndroidManifest.xml:28:13-35
        is also present at [com.android.support:exifinterface:26.0.1] AndroidManifest.xml:25:13-35 value=(26.0.1).
     Suggestion: add 'tools:replace="android:value"' to <meta-data> element at AndroidManifest.xml:26:9-28:38 to override.

:(
```

nicemanis, 2018-01-17:
@daFelix ahh here is the problem -> `com.android.support:appcompat-v7:26.1.0` this dependency uses 26.1.0, but you have to set it to use 26.0.1

although, the strange thing is that I see that in the `build.gradle` file you're compiling 26.0.1 version `compile 'com.android.support:appcompat-v7:26.0.1'` can you confirm that this is true? Are there any warnings or errors during the build that indicate that it will use 26.1.0 instead of 26.0.1? Also you can check you `node_modules` directory, search all build.gradle files for `com.android.support` maybe there are some other packages that depend on this, if so, then they should be compiled in the same manner as in the example i.e. by excluding the appropriate group and compiling 26.0.1 version

daFelix, 2018-01-18:
@nicemanis yeaa i can confirm that in my dependencies there is no `'com.android.support:appcompat-v7:26.1.0'` and I have searched every file in node_module that there is nothing that it use 26.1.0.

Is there another place inside the project that I can keep searching for that file ?

Tsiniloiv, 2018-01-18:
I am having this same issue.

```
Error:Execution failed for task ':react-native-camera:processDebugAndroidTestManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at manifestMerger5515421130866798273.xml:23:9-25:38 to override.
```

Here's my AndroidManifest.xml

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools"
    package="com.kulascopenative"
    android:versionCode="1"
    android:versionName="1.0">`

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

    <uses-sdk
        android:minSdkVersion="16"
        android:targetSdkVersion="22" />

    <application
      android:name=".MainApplication"
      android:allowBackup="true"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:theme="@style/AppTheme"
        tools:node="replace">
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
        android:windowSoftInputMode="adjustResize"
          android:screenOrientation="sensorLandscape">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />
    </application>

</manifest>
```

And here's my build.gradle file:

```
apply plugin: "com.android.application"

import com.android.build.OutputFile

project.ext.react = [
    entryFile: "index.js"
]

apply from: "../../node_modules/react-native/react.gradle"

def enableSeparateBuildPerCPUArchitecture = false

def enableProguardInReleaseBuilds = false

android {
    compileSdkVersion 23
    buildToolsVersion "23.0.1"

    defaultConfig {
        applicationId "com.kulascopenative"
        minSdkVersion 16
        targetSdkVersion 22
        versionCode 1
        versionName "1.0"
        ndk {
            abiFilters "armeabi-v7a", "x86"
        }
    }
    splits {
        abi {
            reset()
            enable enableSeparateBuildPerCPUArchitecture
            universalApk false  // If true, also generate a universal APK
            include "armeabi-v7a", "x86"
        }
    }
    buildTypes {
        release {
            minifyEnabled enableProguardInReleaseBuilds
            proguardFiles getDefaultProguardFile("proguard-android.txt"), "proguard-rules.pro"
        }
    }
    // applicationVariants are e.g. debug, release
    applicationVariants.all { variant ->
        variant.outputs.each { output ->
            // For each separate APK per architecture, set a unique version code as described here:
            // http://tools.android.com/tech-docs/new-build-system/user-guide/apk-splits
            def versionCodes = ["armeabi-v7a":1, "x86":2]
            def abi = output.getFilter(OutputFile.ABI)
            if (abi != null) {  // null for the universal-debug, universal-release variants
                output.versionCodeOverride =
                        versionCodes.get(abi) * 1048576 + defaultConfig.versionCode
            }
        }
    }
}

dependencies {
    compile (project(':react-native-camera')) {
        exclude group: "com.android.support"
    }
    compile ('com.android.support:exifinterface:26.0.1') {
        force = true;
    }
    compile project(':react-native-orientation')
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
}

task copyDownloadableDepsToLibs(type: Copy) {
    from configurations.compile
    into 'libs'
}
```

nicemanis, 2018-01-18:
@Tsiniloiv change the version for exifinterface from 26.0.1 to 26.0.2 here:
```
compile ('com.android.support:exifinterface:26.0.2') {
    force = true;
}
```

aamirmsw, 2018-01-18:
@nicemanis I am facing the same problem  **Error:Execution failed for task ':react-native-camera:processDebugAndroidTestManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at manifestMerger1929398336371439706.xml:23:9-25:38 to override.** Tried all the above solutions but still getting error.

nicemanis, 2018-01-18:
@aamirmsw the same issue
change this:
```
compile ('com.android.support:exifinterface:26.0.1') {
    force = true;
}
```
to this:
```
compile ('com.android.support:exifinterface:26.0.2') {
    force = true;
}
```

nicemanis, 2018-01-18:
@aamirmsw @Tsiniloiv have you added this to your `android/app/build.gradle` ?

```Gradle
dependencies {
    compile (project(':react-native-camera')) {
        exclude group: "com.android.support"

        // uncomment this if also com.google.android.gms:play-services-vision versions are conflicting
        // this can happen if you use react-native-firebase
        // exclude group: "com.google.android.gms"
    }

    compile ('com.android.support:exifinterface:26.0.1') {
        force = true;
    }

    // uncomment this if you uncommented the previous line
    // compile ('com.google.android.gms:play-services-vision:11.6.0') {
    //    force = true;
    // }
}
```

aamirmsw, 2018-01-18:
@nicemanis Yes ,i have added the same but issue still persist.

Tsiniloiv, 2018-01-18:
I've added those parts and updated the exifinterface compile version number. It seems to occasionally throw the error, and there's a redline at the compile command for appcompat, but when it starts throwing that error rebuilding the whole project seems to work.

Rewieer, 2018-01-19:
I had the same problem. 
In the README.md it is said to add : 
`xmlns:tools="http://schemas.android.com/tools"` and `tools:node="replace"`

Actually adding these lines were the problem. When I removed them it was all working back again.

jgcmarins, 2018-01-19:
@Rewieer which version?

Rewieer, 2018-01-19:
@jgcmarins react-native-camera@0.13.0, react-native@0.52.0, react-native-firebase@3.2.2

lynkxyz, 2018-01-19:
Same error here

muratozgul, 2018-01-19:
Same problem, react-native-camera@0.13.0, react-native@0.51.0

Barylskyigb, 2018-01-20:
Problem was solved for me after I installed react-native-camera@0.6.0 instead of v0.13.0. react-native version in 0.47.2

JaHeRoth, 2018-01-22:
`react-native-camera@0.6.0` solved this problem, but would crash the app if I tried to actually import the module.

What worked fully for me was using `react-native-camera@0.10.0` instead.

In other words, the full command from start to finish became `react-native init YOUR_PROJECT_NAME; cd YOUR_PROJECT_NAME; yarn add react-native-camera@0.10.0; react-native link`

AkshayKomarla, 2018-01-24:
i'm still getting this same issue. using master branch

`> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.github.react-native-community:cameraview:df60b07573.
     Required by:
         project :app > project :react-native-camera`

tekslave, 2018-01-24:
**Add** jitpack to android/build.gradle
`allprojects {
    repositories {
        maven { url "https://jitpack.io" }
    }
}`

- you would see other repositories there just add **maven {url "https://jitpack.io"}**

-  now in android/app/build.gradle  in the dependencies add
`compile (project(':react-native-camera'))`

AkshayKomarla, 2018-01-25:
its already added that part of code

pihyper, 2018-01-25:
I was able to solve this by following the steps at

https://github.com/react-native-community/react-native-camera#meta-data-android-26

poollau, 2018-01-26:
I was able to slove the error as what @JaHeRoth commented.
Just use the react-native-camera@0.12.0
seems there is something wrong with the lastest version~

ddresch, 2018-01-29:
For some blind guys like me. There are **two** `build.gradle` files in your project. Like @tekslave wrote one is located at `android/build.gradle` and the other at `android/app/build.gradle`. So all those hints about **jitpack.io** reference the first one!

18913132362, 2018-02-01:
I also met with the same problem as the landlord who solved the problem and I needed help

frostless, 2018-02-04:
I believe I tried almost all the above solutions but the issue still persist;

1: I use the master version of the camera
2:I added maven { url "https://jitpack.io" } to the android buildscript repositories
3:I changed android/app/build.gradle  according to @nicemanis and I am not using firebase.

Error messages:

> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.

Here is my package.json
{
  "name": "",
  "version": "0.1.0",
  "private": true,
  "devDependencies": {
    "babel-preset-react-native-stage-0": "^1.0.1",
    "jest-react-native": "^18.0.0",
    "react-test-renderer": "16.2.0"
  },
  "scripts": {
    "start": "react-native start",
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "test": "node node_modules/jest/bin/jest.js"
  },
  "jest": {
    "preset": "react-native"
  },
  "dependencies": {
    "react": "16.2.0",
    "react-native": "0.52.0",
    "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera",
    "react-native-modal-dropdown": "^0.6.0",
    "react-native-svg": "6.0.1-rc.1",
    "react-native-svg-charts": "^2.2.1",
    "react-native-tab-view": "0.0.74",
    "react-navigation": "^1.0.0-beta.27"
  }
}

And build.gradle in android/app

dependencies {
    compile project(':react-native-svg')
    // compile project(':react-native-camera')
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
  compile (project(':react-native-camera')) {
        exclude group: "com.android.support"

        // uncomment this if also com.google.android.gms:play-services-vision versions are conflicting
        // this can happen if you use react-native-firebase
        // exclude group: "com.google.android.gms"
    }

    compile ('com.android.support:exifinterface:26.0.1') {
        force = true;
    }

    // uncomment this if you uncommented the previous line
    // compile ('com.google.android.gms:play-services-vision:11.6.0') {
    //    force = true;
    // }
    
}

build.gradle in android

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
            url "https://jitpack.io"
            url "https://maven.google.com"
        }
    }
}

finally the manifest

<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.buyingagent"
    android:versionCode="1"
    android:versionName="1.0">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.RECORD_AUDIO"/>
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

    <uses-sdk
        android:minSdkVersion="16"
        android:targetSdkVersion="22" />

    <application
      android:name=".MainApplication"
      android:allowBackup="true"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
        android:windowSoftInputMode="adjustResize">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />
    </application>

</manifest>

pihyper, 2018-02-04:
@frostless : Try these steps

frostless, 2018-02-04:
@pihyper  I changed the setting but the the issue still persist... 
That is actually my first time build the app in android, I debugged and tested in iOS along the way.
But in android everything seems broken..

chenzxl, 2018-02-07:
@AkshayKomarla,I have the same problem with you. How do you solve it

chenzxl, 2018-02-07:
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.
     Required by:
         jiyingApp:app:unspecified > jiyingApp:react-native-camera:unspecified

* Try:

ysq1051838264, 2018-02-08:
Error:Failed to resolve: com.github.react-native-community:cameraview:df60b07573

johnnymm, 2018-03-02:
Downgrade to 0.12.0

react-native unlink react-native-camera
npm install --save react-native-camera@0.12.0
react-native link react-native-camera

johnnymm, 2018-03-02:
the gradle files is not updated when upgrade to 1.0.0
https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md

After following the upgrade guide, I need to open project in android studio to resolve further dependencies.

android/app/build.gradle

dependencies {
    compile (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
    }
    compile ("com.google.android.gms:play-services-vision:10.2.0") {
        force = true;
    }
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
}

android/build.gradle

buildscript {
    repositories {
        jcenter()
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.2.3'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven { url "https://jitpack.io" }
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
    }
}

Now it is able to compile and run.

twitchett, 2018-03-11:
Thanks @johnnymm, this helped me.

pollaro, 2018-03-13:
Great job, @johnnymm

MrZhang123, 2018-03-15:
@johnnymm Thanks a lot , this helped me too