react-native-camera/react-native-camera #1089: Android cropToPreview only cropping left/right sides

sunweiyang, 2018-01-10:
This is related to #1057.

I am running RN 0.51.0 and react-native-camera master as of bc975b4bac49a31b8ff885efddd36c456da7c262.

On iOS it works great, but on Android I'm seeing that it only crops the left/right sides off. In other words, I've defined my preview to be a square, but what I get instead is a tall skinny rectangular captured image.

I have the following constant defined:
```
const windowWidth = Dimensions.get('window').width;
```

A button is defined that runs this function when pressed (very similar to README example):
```
  takePicture() {
    var component = this;
    const options = {};
    //options.location = ...
    this.camera.capture({metadata: options})
      .then((data) => {
        console.log(data);
      })
      .catch(err => console.error(err));
  }
```

My render method contains:
```
        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={{width: windowWidth*0.8, height: windowWidth*0.8}}
          type={'front'}
          orientation={'portrait'}
          cropToPreview={true}
        />
```

mckn, 2018-01-11:
@sunweiyang awesome feedback! Thanks! I will try to setup your example and fix the issue.

mckn, 2018-01-11:
@sunweiyang Some more questions so I can debug.

1. What orientation settings do you have on your app? 
2. What orientation do you have when taking the photo?

Thanks in advance!

sunweiyang, 2018-01-12:
Thanks @mckn for looking into this.

I am forcing portrait mode on my app. Specifically, I have the following in my AndroidManifest.xml:
```
        <activity android:name=".MainActivity"
            android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
            android:windowSoftInputMode="adjustPan"
            android:exported="true"
            android:screenOrientation="portrait" />
```

claudiumatei, 2018-02-06:
Hi, I also confirm the issue on Android (no matter the orientation of the device).

eoy, 2018-02-21:
I can also confirm this issue on Android.

t407289575, 2018-03-02:
Why the 'cropToPreview={true}'  is invalid for me ,both Android or IOS.

    "react-native": "0.48.4",
    "react-native-camera": "^0.13.0",

my code:
<Camera
                    ref={(cam) => {
                        this.camera = cam;
                    }}
                    style={{
                        width:SCREEN_WIDTH * 0.8,
                        height: SCREEN_WIDTH * 0.5,
                    }}
                    captureTarget={Camera.constants.CaptureTarget.cameraRoll}
                    aspect={Camera.constants.Aspect.fill}
                    captureQuality={'high'}
                    cropToPreview={true}
                    playSoundOnCapture={false}>

Hope to help

Jinran, 2018-05-07:
does this issue fixed？

eqyiel, 2018-06-15:
Ref: https://github.com/react-native-community/react-native-camera/pull/1636

closed 2019-06-08 by stale[bot]