react-native-camera/react-native-camera #1090: Front camera not working on some iPhones

tommynordli, 2018-01-11:
I'm having issues using the front facing camera on some iPhones, at least on iPhone 5s and 6S, both running iOS 11. I have no issues on other iPhones, e.g. iPhone X. The preview is black, and the app crashes when I try to take a photo. 

[image]
_Trying to use the front facing camera on iPhone 5s_

VrajSolanki, 2018-01-31:
Same Issue for me but removing the capture quality and setting default mode to front doesn't work for me.
device: iPhone 6

sibelius, 2018-01-31:
Is this happening to RNCamera as well?

VrajSolanki, 2018-01-31:
@sibelius  fixed it with removing capture quality.

sibelius, 2018-01-31:
Can somebody send a fix for this?

nadzic, 2018-04-15:
I am using react-native-camera, tried to run following code https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md in the simulator and Iphone X (iOS 11.2) and getting just black screen (as it is set background) without any camera. 

Any issues related to this?

IssuehuntBot, 2018-11-19:
@0maxxam0 funded this issue with $20. [See it on IssueHunt](https://issuehunt.io/repos/33218414/issues/1090)

sibelius, 2019-04-18:
RCTCamera is removed https://github.com/react-native-community/react-native-camera/issues/1890

Migrate to latest version of RNCamera

closed 2019-04-18 by sibelius

rezasazesh, 2019-07-29:
@sibelius this still happens on 3.0.0

joebernard, 2021-07-19:
This is still an issue on `3.44.1`.