react-native-camera/react-native-camera #1091: Error creating media file

aspidvip, 2018-01-12:
Hello!
React-Native: 0.49.5
react-native-camera: 0.12.0
Android: 4.2.2
compileSdkVersion 26
buildToolsVersion '26.0.1' || buildToolsVersion '26.0.2'

```
<Camera
            ref={(cam) => {
              this.camera = cam;
            }}
            style={styles.preview}
            aspect={Camera.constants.Aspect.fill}
            cropToPreview={true}
            captureTarget={Camera.constants.CaptureTarget.cameraRoll}
            orientation={Camera.constants.Orientation.auto}
            flashMode={Camera.constants.FlashMode.auto}
            defaultTouchToFocus
            mirrorImage={false}
            ></Camera>
```

[image]

sibelius, 2018-01-12:
Please provide full stacktrace

Try with master version as well

aspidvip, 2018-01-12:
`I/dalvikvm-heap: Grow heap (frag case) to 8.442MB for 1228812-byte allocation
D/dalvikvm: GC_FOR_ALLOC freed 0K, 21% free 8530K/10700K, paused 3ms, total 3ms
E/RCTCameraModule: failed to create directory:/mnt/sdcard/DCIM
E/ReactNativeJS: { [Error: Error creating media file.] framesToPop: 1, code: 'EUNSPECIFIED' }
I/Choreographer: Skipped 32 frames!  The application may be doing too much work on its main thread.
`

jgcmarins, 2018-01-12:
Hello @aspidvip ,
I saw that you are running on:

> react-native-camera: 0.12.0

Could you try with master version, while the release version isn't updated?
Thanks.

aspidvip, 2018-01-12:
how can i update it

jgcmarins, 2018-01-12:
add this to your package.json: `"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera",`

aspidvip, 2018-01-13:
Error:Failed to resolve: com.github.react-native-community:cameraview:df60b07573
<a href="openFile:/Users/aspid/Desktop/Publikun/android/app/build.gradle">Open File</a><br><a href="open.dependency.in.project.structure">Show in Project Structure dialog</a>

sibelius, 2018-01-13:
You need jitpack, check README

cainaleaouk, 2018-01-31:
Hi there,

I am having the same issue. I have all the same setup as the original post. 

Here is the log:
```01-31 13:10:15.177 30716-30716/com.skeletonapp E/MediaRecorder: start failed: -38
01-31 13:10:15.197 30716-30716/com.skeletonapp E/RCTCameraModule: Media recorder start error.
                                                                  java.lang.IllegalStateException
                                                                      at android.media.MediaRecorder.start(Native Method)
                                                                      at com.lwansbrough.RCTCamera.RCTCameraModule.record(RCTCameraModule.java:357)
                                                                      at com.lwansbrough.RCTCamera.RCTCameraModule.captureWithOrientation(RCTCameraModule.java:514)
                                                                      at com.lwansbrough.RCTCamera.RCTCameraModule.access$100(RCTCameraModule.java:38)
                                                                      at com.lwansbrough.RCTCamera.RCTCameraModule$2.orientationEvent(RCTCameraModule.java:498)
                                                                      at com.lwansbrough.RCTCamera.RCTSensorOrientationChecker$Listener.onSensorChanged(RCTSensorOrientationChecker.java:64)
                                                                      at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:630)
                                                                      at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                      at android.os.MessageQueue.next(MessageQueue.java:323)
                                                                      at android.os.Looper.loop(Looper.java:138)
                                                                      at android.app.ActivityThread.main(ActivityThread.java:5454)
                                                                      at java.lang.reflect.Method.invoke(Native Method)
                                                                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
01-31 13:10:15.207 30716-30758/com.skeletonapp E/ReactNativeJS: { [Error: null] framesToPop: 1, code: 'EUNSPECIFIED' }```

I also have `jitpack` added to my build.gradle:
```allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            url "https://jitpack.io"
        }
    }
}
```

I can take pictures just fine, when I try to record a video I get that crash on AndroidStudio and in the app the message is just `null` from `<unknown>`.

Any ideas what could it be?

n1ru4l, 2018-12-12:
`RCTCamera` is deprecated. Please try to use `RNCamera`instead.

closed 2018-12-12 by n1ru4l