react-native-camera/react-native-camera #1093: Feature Request: Configure destination folder?

hardcodet, 2018-01-12:
I think with a custom camera, it would be a rather common requirement to actually process an image after capture, or store it within a dedicated folder of the app for later reuse.

The API currently gives us abstractions (Camera Roll, Disk, Temp), but being able to override those with a specified file path would be great (and hopefully not much of an issue, but then, I'm very new to React/native development).

From the top of my head, this would ensure:
* no overhead with moving files around
* easy deletion of temp files, even after an app crash
* confidentiality, where needed
* Simple file access from other areas of the application without being aware of directories outside the app's root storage
* ...

ospfranco, 2018-01-12:
While I think this is technically feasible on android, iOS on the other hand is very restrictive on the file system, read:

https://developer.apple.com/library/content/documentation/FileManagement/Conceptual/FileSystemProgrammingGuide/FileSystemOverview/FileSystemOverview.html

hardcodet, 2018-01-12:
@ospfranco that was quick feedback :)

Following the link, wouldn't that be the Documents folder my app has easy access to?

sibelius, 2018-01-12:
use https://github.com/itinance/react-native-fs to move files to another place

let's keep this package simpler

closed 2018-01-12 by sibelius

hardcodet, 2018-01-12:
@sibelius Of course I'll accept that - thanks for the quick response time 👍 

I disagree though on this making the package more complex - I would think that storing files after capturing them is a quintessential feature. The abstractions you provide make sense, especially for the camera roll, but I'd guess that most apps that use the "disk" or "temp" options would have to jump through the hoops of dealing with extra packages just to get the files to their designated destinations.

Just my 0.02$ - thanks for all the work you've put in so far. The package rocks :)